rgleason/weatherfax_pi #48: Excessive runtime linux deps: rtlsdr

leamas, 2021-01-13:
The plugin seemingly depends on librtlsdr, the Debian packaging of https://github.com/osmocom/rtl-sdr.  This causes problems with the new plugin installer since users manually must install the Debian package for plugin to work.

The first action would be if this software actually uses the dependency. If so, it might be possible to rebuild it from source, or find some other solution to walk around the problem.

rgleason, 2021-04-22:
Alec
I hope to be updating weatherfax data files soon and need to know what to do about this. 

Linux users and RPI have been using the rtlsdr dongle to download weatherfax into Opencpn and apparently its been working for some time now. 

The advantage is that expensive weather services can be avoided completely.  I have not been successful getting an rtlsdr dongle to work with weatherfax_pi in Windows, but would very much like to get it working.

There are various workarounds that may not be quite as convenient.

So you want us to remove librtlsdr.... is there a workaround of some sort?  I will need to discuss this with Sean too.

leamas, 2021-04-22:
> So you want us to remove librtlsdr.... 

Certainly not. What we need to do is to remove dependencies on external packages, more or less as on Windows or MacOS

The solution might be to bundle it on linux, that seems like the simplest solution there. See for example photolayer which does this for linux deps.

leamas, 2021-04-22:
hm... or perhaps it might be better to rebuild it from source, that should solve the situation at least also for windows. The package doesn't seem that complicated to build, see https://osmocom.org/projects/rtl-sdr/repository

leamas, 2021-04-22:
A quick check: library builds out of the box on both windows and linux. Probably the best path, fixes problems for mulitple platforms (or at least a chance).

leamas, 2021-04-22:
It does not build out of the box on MacOS. First,  libusb needs to be installed using brew. There seems to be a bug in CMakeLists as well, it locates libusb but does not include the linker flags.  Probably simple to fix.

rgleason, 2021-04-22:
Thank you Alec, I did not realize you were actually building!  There might  be a way to support rtlsdr with a little additional work.

leamas, 2021-04-23:
I just made some quick checks. No time to push this forward ...

rgleason, 2024-12-22:
I do not know if this is fixed.  For more detail about PhotoLayer  fixes  see  
https://github.com/OpenCPN/OpenCPN/issues/2149#issuecomment-761869946

bdbcat, 2024-12-23:
Close it.  All this is 4 years old, and has been OBE (Overtaken By Events).

rgleason, 2024-12-23:
Thanks Dave. Closing

closed 2024-12-23 by rgleason