rgleason/weatherfax_pi #50: Mapping Failed

ThomTrab, 2021-04-21:
Hello,

I start using this plugin some months ago without issue.
Now when I try to download fax images from Internet from German Weather DWD or UK MET for example. I got the following error message.
[image]

What can I do to solve this ? I already try to reinstall the plugin.

Thank you for your help

rgleason, 2021-04-21:
Tom, the underlying website has changed so the data files need to be updated. Do you know what the new url is?  There is more info in the manual.

ThomTrab, 2021-04-21:
Thanks for your answer. But I'm not sure to follow you.

After the above error message, I get the following window, that ask me to adapt the image to the map. This part was automatic, as far as I remember.
[image]
That let me think that download is OK, isn't it ?

> There is more info in the manual.

Do you mean edit the `WeatherFaxInternetRetrieval.xml`file ?

ThomTrab, 2021-04-21:
I don't know how but I had messed up with my `CoordinateSets.xml`.

Copying over the one from the git repo solved my issue. :+1:

I apologize for the disturbance.

closed 2021-04-21 by ThomTrab

rgleason, 2021-04-21:
Dear Thom
You can learn how to use the Image Wizard to adjust any image to overlay properly onto Opencpn charts.
The manual should give you some ideas how to do that.
Those settings are then saved in coordinatedatasets.xml  and weatherfaxinternetretrieval.xml

We need people who are familiar with this process and can help update the files.

rgleason, 2021-04-22:
Thank you ThomTrab.  That will be very helpful.  Between me, you and Hans who also is good at this, we should be able to get this done reasonably quickly. I will contact you and Hans with a workflow proposal to split up the areas.  I really like your PR, that is the best way to do this.