rgleason/weatherfax_pi #52: Regarding User writable data

rgleason, 2021-05-10:
Filochard reports:

There is no problem for weatherfax : it creates the good path to store user's files inside flatpak
/home/yourusername/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/weatherfax/
where you may find the downloaded faxes (.png or .tif)
and the .xml files (WeatherFaxInternetRetrieval.xml WeatherFaxSchedules.xml and CoordinateSets.xml)

rgleason, 2021-05-28:
@filochard
Can you test branch master-update
V1.9.14.20
found in cloudsmith?

filochard, 2021-05-28:
Hi Rick
I never had any problem with weatherfax

I downloaded this new version, 
installed it by updating the previous plugin, 
tested it (partially : no HF connected, only internet)
It works perfectly with "German Weather DWD" or "Weathercharts UK"
The layer applies in the right place with the right transparency

the .xml files (WeatherFaxInternetRetrieval.xml WeatherFaxSchedules.xml and CoordinateSets.xml)
and the png or gif files are always stored in the right directory  : 
/home/myusername/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/weatherfax/

PS the same package version is in 
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/

filochard, 2021-05-28:
I tested it with flatpak : 
I don't use Ubuntu but Mageia, not yet allowing Plugin Manager (thinking about setting up a CI build chains which may build the plugins for Mageia)

rgleason, 2021-05-28:
Thanks filochard.   Yes, I recall the discussion with Alec about Mageia.  
I am going to try to get Noaa OPC and OPC working next, they seem to be failing.

closed 2021-05-28 by rgleason