rgleason/weatherfax_pi #53: Export to .kap corrupted

milknuts, 2021-05-18:
Export to .kap produces the file which cannot be opened by OpenCPN

opencpn.log:

23:42:45.377 MESSAGE chartdb.cpp:1346 Creating new chart
23:42:45.378 MESSAGE chartdb.cpp:1475 Initializing Chart /home/denis/Charts/custom/AR_Comodoro_Rivadavida.kap
23:42:45.378 MESSAGE chartimg.cpp:1673    Invalid nColorSize data, corrupt on chart /home/denis/Charts/custom/AR_Comodoro_Rivadavida.kap
23:42:45.378 MESSAGE chartdb.cpp:1585 Problem initializing Chart /home/denis/Charts/custom/AR_Comodoro_Rivadavida.kap
23:42:45.378 MESSAGE chartdb.cpp:1605    OpenChartFromStack... Error opening chart /home/denis/Charts/custom/AR_Comodoro_Rivadavida.kap ... return code 2

rgleason, 2022-05-03:
I can confirm that Export to Kap is not working and needs to be fixed.
When the files are moved to a Chart-Kap directory, the db is scanned, and a Chart Group made with the chart-kap path, when this group is selected the other kaps show ok, but the weatherfax kaps do not.

rgleason, 2022-05-03:
When I open a wf generated file with a text editor I get this header, for example

```
! 2013 opencpn WeatherFax 1.9 file generator
! Map created at 03/05/2022
! Size in milles 4858.69 x 3843.82
! Resolution units METERS - 4389.41x4229.80 -> 42298009 at 254 dpi
VER/3.0
OST/1
CED/SE=1,RE=1,ED=03/05/2022
BSB/NA=German Weather DWD - European - B
    NU=UNKNOWN,RA=2050,1683,DU=254
KNP/SC=42298009,GD=WGS 84,PR=MERCATOR,PP=0.00
    PI=UNKNOWN,SP=UNKNOWN,SK=0.0,TA=90
    UN=METERS,SD=,DX=4389.41,DY=4229.80
REF/1,1084,509,70.000000,10.000000
REF/2,664,509,70.000000,-20.000000
REF/3,664,1459,30.000000,-20.000000
REF/4,1084,1459,30.000000,10.000000
PLY/1,79.328703,-67.428571
PLY/2,79.328703,79.000000
PLY/3,15.265034,79.000000
PLY/4,15.265034,-67.428571
DTM/0.000000,0.000000
IFM/1
RGB/0,35,41,37
RGB/1,118,127,192
```
With the image file below.
[German Weather DWD - European - B.zip](https://github.com/rgleason/weatherfax_pi/files/8610973/German.Weather.DWD.-.European.-.B.zip)

Here is another example
[NOAA - Boston - 8.zip](https://github.com/rgleason/weatherfax_pi/files/8610970/NOAA.-.Boston.-.8.zip)

rgleason, 2024-12-20:
How can we get this fixed?

rgleason, 2024-12-23:
I exported  a NOAA Boston weatherfax file  using File > Export.
I took that file and put it in MBtiles folder I had created under Chart dir
I added the path to Charts, make a Group call MBtiles and used the MBtile path.
Then I tried to display it as an overlay onto charts. This is the result.
Therefore:  OpenCPN Weatherfax_pi does make good KAP files.

[image]

Here it the file created if you want to try it.
[NOAA - Boston - 4.zip](https://github.com/user-attachments/files/18224592/NOAA.-.Boston.-.4.zip)

closed 2024-12-23 by rgleason