rgleason/weatherfax_pi #74: Bug: Retrieve > Update data fails

rgleason, 2022-05-03:
This function does not work with Sean's data directory.
I changed it to my repository  "data" and it also fails with
[image]

This needs to be fixed.

rgleason, 2022-05-03:
Regarding "Retrieve &gt; Update Data"
I tried changing the url to
```
to: "https://github.com/rgleason/weatherfax_pi/blob/master/data/") ); 
from: "https://raw.githubusercontent.com/rgleason/weatherfax_pi/master/data/") );

```
 still get the same error so I think something has changed and is not coded properly anymore.
[image]

SCarns, 2022-05-05:
Confirm that this happens on Mac OS X Catalina, also.

ThomTrab, 2022-05-07:
Dear all, 

I also confirmed the issue.

I investigated it quickly this morning. Apparently GitHub change the way how to access direct file from a repo.

The hardcoded link `https://raw.githubusercontent.com/seandepagnier/weatherfax_pi/master/data/` in the `weatherfax_pi.cpp` line 306, is not available anymore.

From test, it seems, it need to be replaced by `https://github.com/seandepagnier/weatherfax_pi/blob/master/data/`

I'm not that proeficient with Github and don't have more time this weekend to take care of the modification and open a PR for it. If nobody does it before, I will try to do this begining of next week.

Regards,

rgleason, 2022-05-07:
Tom Trab  thanks for the report. I had discovered that too and changed the url to my repos with that URL, and it does not work because it is still just expecting a single file for weatherfaxInternetRetrieval.xml rather than 12 separate ones.  I may figure it out in some time.

rgleason, 2022-05-09:
Some searches to determine how  WeatherfaxInternetRetrieval.xml and CoordinateSets.xml were downloaded.
Searches include "update data", OCPN_downloadFile, download, downloadthread, WeatherfaxInternetRetrieval.xml, CoordinateSets.xml

[WxFax-update-data-sources.txt](https://github.com/rgleason/weatherfax_pi/files/8653187/WxFax-update-data-sources.txt)

rgleason, 2022-05-09:
The file Weatherfax.cpp  starting at line 831 needs to be augmented to iterate all the files above.

```
void WeatherFax::OnUpdateData( wxCommandEvent& event )
{
    if( DownloadFile( _T("WeatherFaxInternetRetrieval.xml") ) &&
       DownloadFile( _T("CoordinateSets.xml") ) ) {
        m_InternetRetrievalDialog.Load(true);
    }
    if( DownloadFile( _T("WeatherFaxSchedules.xml") ) ) {
        m_SchedulesDialog.Load(true);
    }

}
```

Not quite sure how to do this.

rgleason, 2022-05-10:
Thomas Traub has provided a PR which has been merged, which appears to solve the Retrieve > Update Data feature.
I have directed the requests to the rgleason/weatherfax_pi/data directory until Sean merges my PR, because we have to have access to a current data directory.

I think we should have a message after completion, which appears,
"Data files updated successfully" for some seconds.
Similar if it is not successful.

rgleason, 2022-05-15:
I am going to close this issue now.

closed 2022-05-15 by rgleason