rgleason/weatherfax_pi #76: Download URLs UKMet outdated

ManfredRad, 2022-05-04:
UKMet is downloaded from http://old.wetterzentrale.de/pics/. The server does not exist any more. The info in WeatherFaxInternetRetrieval_Europe.xml should read like below.
I also suggest to rename the region from central to Europe, since the area covered is all of Europe plus some of the atlantic.

 <Server Name="UK MET" Url="https://www.wetterzentrale.de/maps/">
    <Region Name="Europe">
      <Map Url="UKMHDOPEU00_0_1.png" Contents="Surface Anlys 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_6_1.png" Contents="6hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_12_1.png" Contents="12hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_18_1.png" Contents="18hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_24_1.png" Contents="24hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_30_1.png" Contents="30hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_36_1.png" Contents="36hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_42_1.png" Contents="42hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_48_1.png" Contents="48hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_54_1.png" Contents="54hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_60_1.png" Contents="60hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_66_1.png" Contents="66hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_72_1.png" Contents="72hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_78_1.png" Contents="78hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_84_1.png" Contents="84hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_90_1.png" Contents="90hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_96_1.png" Contents="96hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_102_1.png" Contents="102hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_108_1.png" Contents="108hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_114_1.png" Contents="114hr Forecast 00Z" Area="A" />
      <Map Url="UKMHDOPEU00_120_1.png" Contents="120hr Forecast 00Z" Area="A" />
      <Area Name="A" lat1="40N" lat2="80N" lon1="40E" lon2="60W" />
    </Region>
  </Server>

rgleason, 2022-05-05:
The link above does not work for me, but this does seem to work https://www.wetterzentrale.de/
I am not sure how much work is involved to fix this, and I do not have time to do it.
We expect that users in the region will update the appropriate files and make a PR or send the files.

ManfredRad, 2022-05-05:
Here is the file, which needs to be changed to access UKMet data. It has been given the extension .txt to upload it here. The extension .txt needs to be removed and the file copied to data, if tested and deemed working.

[WeatherFaxInternetRetrieval_Europe.xml.txt](https://github.com/rgleason/weatherfax_pi/files/8631042/WeatherFaxInternetRetrieval_Europe.xml.txt)

rgleason, 2022-05-05:
Good job. Have updated WFIR_European  and committed v1.9.30.4 and pretty soon after the cloud builds the two commits you will find your tarballs in the Cloudsmith Opencpn  weatherfax-beta repository (this link is in the spreadsheet at the top) Look at the linked spreadsheet headers, One is for Cloudsmith O Beta and the other next one down is for the plugin's github repos.
You can use "Import Plugin" after downloading your tarballs for the OS you need.

Later I will update TP frontend2 with Jon's current work and push up to the beta catalog.

Someday, we can go through how to make a PR. The first time it is more invovled, after that it gets better.

ManfredRad, 2022-05-06:
Thanks. Install of the new version worked, however there was an error in the xml file I uploaded. I will fix that and extent the WFIR_European to make use of the current state of the art models (ECWMF and Icon). In my opinion UK_MET is obsolete here, since it is actually based on the other European models anyhow.

rgleason, 2022-05-07:
I am finding it difficult to use these images at  https://www.wetterzentrale.de/
because they do not have Long/Lat coordinates.  The only way to georeference them is to determine two landmarks beforehand, record the lat/long and then use Image Wizard by selecting those landmarks and entering the coords.

It can be done, and it is tedious, but possible.

ManfredRad, 2022-05-07:
I believe, we do not need them. Most of them are just displays of griffiges anyhow. If we want to go with synoptic weather charts only, then we should not use them. One source is the german weather service DWD. And those are working. A - potentially useful - source for the British met office is [here](https://www.metoffice.gov.uk/weather/maps-and-charts/surface-pressure):  

These are great maps for the north Atlantic and they could be helpful for all on the way from US/Carib to Europe.

They even have b&w. Unfortunately the URLs to the individual map has a time stamp in the path. Like: https://data.consumer-digital.api.metoffice.gov.uk/v1/surface-pressure/bw/2022-05-07T0000/0000_MEDIUM_RANGE_FC036.gif

I do not know, if WeatherFax can deal with this.

rgleason, 2022-05-09:
@ozolli  I think what you are doing is going to be good.  We won't need the B&W probably because WxchartsUK has them.

ozolli, 2022-05-09:
> I am finding it difficult to use these images at https://www.wetterzentrale.de/
> because they do not have Long/Lat coordinates. The only way to georeference them is to determine two landmarks [...]

I use the lat/lon crossings near some known places to determine their coordinates as they are seen as dotted lines in the chart, i.e. here Svaldbard/Portugal crossing for the red coords 80N/9W and Greenland/Africa just South of Mediterranean Sea for the green coords 30N/40W.

ozolli, 2022-05-10:
I just realized this morning that these charts are dated 27APR2022.
There is something wrong...
Same on the website : 
https://www.wetterzentrale.de/de/topkarten.php?map=1&model=ukmhd&var=1&time=0&run=00&lid=OP&h=0&tr=24&mv=0#mapref

rgleason, 2022-05-10:
Maybe we drop those,  GFS has these which might be easier to geo coord 
https://www.wetterzentrale.de/en/topkarten.php?model=gfs&lid=OP
Maybe there is something better to use in here?  Didn't look too hard.

rgleason, 2022-05-10:
If you hover on the NAVGEM , UKMO and UKMO EU  when on 500hPa charts there is a message showing they are not current.

ozolli, 2022-05-10:
In fact UKMO 500HPa geo is up to date !
https://www.wetterzentrale.de/en/topkarten.php?map=1&model=ukm&var=1&time=72&run=0&lid=OP&h=0&tr=24&mv=0

ozolli, 2022-05-10:
ECMWF 500HPa Geopot 0h to 240h step 6h done.

rgleason, 2022-05-10:
Good, but CFS, ECMWF, GEM, GFS, ICON  all offer earlier 0  24 hr and up.  GFS has the furthest out.  ECMWF has 0 24 48 72 which might provide the more current end.

ozolli, 2022-05-10:
Yes why not. I usually get CAPE with grib files, like wind.
I can do this for other people, no problem at all, but I only use images for MSLP analysis and forecast so I have more information than what a grib file can give to me.

Speaking of this, I often make sure how far forecasting can be reliable using the so called spaghetti plate : 
https://www.wetterzentrale.de/en/topkarten.php?map=1&model=gfs&var=1&run=6&time=0&lid=SPAG&h=0&mv=0&tr=24#mapref
This is a comparison of the different parameters computed by the GFS model and how they differ with time.
The longer they "stay together" the more reliable the forecast is.

rgleason, 2022-05-10:
ozolli wrote: Yes why not. I usually get CAPE with grib files, like wind.  - Well that's a good point, but sometimes the weatherfax way is easier when looking at weather.

rgleason, 2022-05-10:
```
Speaking of this, I often make sure how far forecasting can be reliable using the so called spaghetti plate :
https://www.wetterzentrale.de/en/topkarten.php?map=1&model=gfs&var=1&run=6&time=0&lid=SPAG&h=0&mv=0&tr=24#mapref
This is a comparison of the different parameters computed by the GFS model and how they differ with time.
The longer they "stay together" the more reliable the forecast is.
```
Just learned something.  How do you read the spagetti plate though the final one looks really complex.
Now I just stepped through it and understand it looks comprehensible up to about 144 hours.
https://www.wetterzentrale.de/en/topkarten.php?map=1&model=gfs&var=1&run=6&time=144&lid=SPAG&h=0&mv=0&tr=24#mapref

Perhaps that should be in there?

rgleason, 2022-05-15:
@ozolli  I just realized that I may not have gotten the update for this file?  ...or where would I find it?

rgleason, 2022-05-15:
Also the coordinates for this are no longer working.

ozolli, 2022-05-15:
Oh yes I forgot this one... Will do that today.

rgleason, 2022-05-15:
@ozolli OK thanks  very much. Standing by.

ozolli, 2022-05-15:
Well, I worked faster than I thought.
Everything is in this zip: 
[WeatherFax_Update_OZ_15May2022.zip](https://github.com/rgleason/weatherfax_pi/files/8695606/WeatherFax_Update_OZ_15May2022.zip)

rgleason, 2022-05-16:
Ozolli, I've uncompressed and tried these in a new version of weatherfax_pi, and UKMet and WeatherchartsUK are opening in Image Wizard due to missing coordinateSets.xml?
[image]
[image]

The rest of it seems to be working ok.

rgleason, 2022-05-16:
Pushed up to github  https://github.com/rgleason/weatherfax_pi/commits/master
4 flatpak fail + 2 android, but the rest build,   v1.9.30.10
You can find them here in cloudsmith https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.9.30.10

ozolli, 2022-05-16:
> Ozolli, I've uncompressed and tried these in a new version of weatherfax_pi, and UKMet and WeatherchartsUK are opening in Image Wizard due to missing coordinateSets.xml?

Oups... I forgot to include WFIR_Europe.xml in the archive, that's the reason why. Here it is: 
[WeatherFaxInternetRetrieval_Europe.zip](https://github.com/rgleason/weatherfax_pi/files/8697955/WeatherFaxInternetRetrieval_Europe.zip)

Sorry you have to push and compile again...

ozolli, 2022-05-16:
I did a few more mapping refinements (placement and resolution) and some bug fixes (Mostly Bermuda which has a bit weird naming and too short files retention).

Eveything updated is in the following zip archive, including the WFIR_Europe.xml: 
[WeatherFax_Update_OZ_16May2022.zip](https://github.com/rgleason/weatherfax_pi/files/8698946/WeatherFax_Update_OZ_16May2022.zip)

ozolli, 2022-05-16:
I did a few more mapping refinements (placement and resolution) and some bug fixes (Mostly Bermuda which has a bit weird naming and too short files retention).
I also added ICON (in German Weather DWD) and Arpege from Wetterzentrale.

Eveything updated is in the following zip archive, including the WFIR_Europe.xml: 
[WeatherFax_Update_OZ_16May2022.zip](https://github.com/rgleason/weatherfax_pi/files/8700507/WeatherFax_Update_OZ_16May2022.zip)

ozolli, 2022-05-16:
I did a few more mapping refinements (placement and resolution) and some bug fixes (Mostly Bermuda which has a bit weird naming and too short files retention).
I also added ICON (in German Weather DWD), Arome and Arpege from Wetterzentrale.

Eveything updated is in the following zip archive: 
[WeatherFax_Update_OZ_16May2022.zip](https://github.com/rgleason/weatherfax_pi/files/8701140/WeatherFax_Update_OZ_16May2022.zip)

closed 2022-05-17 by rgleason

ozolli, 2022-05-17:
They are in CoordinateSets.xml.
Just checked you commit and Arome is not there. My fault as I have the working CoordinateSets.xml in another folder I must have forgot to sync it before zipping the files.
But Arpege and German Weather DWD are in your commit...
Here is the right file: 
[CoordinateSets.zip](https://github.com/rgleason/weatherfax_pi/files/8708601/CoordinateSets.zip)
There are new NOAA GOES entries as I am including NOAA sat images from here: 
https://www.star.nesdis.noaa.gov/goes/

rgleason, 2022-05-17:
Ozolli  awesome. I will make another commit tonight.
To get an updated version of coordinateSets.xml after you have used Image Manager, are you having to copy your coordiateSets.xml from your  <user writable directory> to the directory that weatherfaxInternetRetrival.xml is installed in?

IE: for Windows10  
copy C:\ProgramData\opencpn\plugins\weatherfax\coordinateSets.xml  
to C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordiateSets.xml  (copy over the old dated one that is there?

Or the equivalent directories for ubuntu/linux?

ozolli, 2022-05-17:
No I don't need to write it elsewhere; It works where it is. I just copy it to where the PIM installs it to make the zip file.
I copy from $HOME/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/
to $HOME/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/weatherfax_pi/

rgleason, 2022-05-18:
> No I don't need to write it elsewhere; It works where it is. I just copy it to where the PIM installs it to make the zip file.
> I copy from $HOME/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/
> to $HOME/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/weatherfax_pi/

Yes, that is what I mean.  It works where it is located, in your user writable directory and it gets changed and saved there as you redefine coordinates with the Image Wizard.  This then makes the installed version in  
$HOME/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/weatherfax_pi/
outdated.

Thanks I think I will use some of this in description in the manual.

ozolli, 2022-05-18:
I cut all files from data/opencpn/plugins/weatherfax_pi/ and pasted them in config/opencpn/plugins/
The plugin works okay except the toolbar images which must stay in data/opencpn/plugins/weatherfax_pi/

rgleason, 2022-05-18:
@ozolli 
Does it look to you like I got all of your work?  See 
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.31.0
for tarballs
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.31.0+tarball

ozolli, 2022-05-18:
Installed it on the RPi, changed the MappingMultiplier for Arpege Europe and everything works fine and seem to be there.

rgleason, 2022-05-18:
Very nice job.