rgleason/weatherfax_pi #77: Cannot Audio Capture

SCarns, 2022-05-05:
Cannot do an audio capture on Mac OS X Catalina. Audio capture sometimes crashes OpenCPN. Preferences dialog won't let me select an input. Also, preferences won't let me choose between "Meters" and "Feet".

SCarns, 2022-05-05:
Also, unable to "Update Sources" to see new locations, such as French Polynesia.

rgleason, 2022-05-05:
What are you using to capture the sounx?  rtlsdr?  Ham Radio? etc?  Please describe your setup.

rgleason, 2022-05-06:
Scarns  I can confirm that Retrieve > Capture and picking a saved sound file does not work.
"Failed to configure capture"

rgleason, 2022-05-07:
Here is a link to some sound files.
https://2bsailing.ca/info/Weatherfax_and_shortwave_radio.php

Sound samples
http://www.jvcomm.de/soundframee.html

SCarns, 2022-05-08:
Sorry Rick. Didn't mean to drop out. I went into the yard this week and my free time evaporated. Thanks for the files, I'll check those out soon and report back.

Steve

rgleason, 2022-05-08:
Steve No problem. Also tryinf to get launched. Lots of peojects.

rgleason, 2024-12-20:
Steve,  I am back at Weatherfax.  Did you see this post? 
https://www.cruisersforum.com/forums/f134/weatherfax-plugin-19-60-0-linx-mint-missing-ui-buttons-278918.html#post3814973
PMX had success downloading sound files, and in this post he describes the whole process
https://www.cruisersforum.com/forums/f134/weatherfax-plugin-19-60-0-linx-mint-missing-ui-buttons-278918.html#post3816899
using an internet connection for the sound files, which makes it easier to test.

Also Francizka has had success retrieving SSB sound files over a radio and she wrote a simple pdf about that that is now in the plugin manual.

Is this issue still valid?

rgleason, 2024-12-20:
PS: WE have a more recent plugin up in Cloudsmith, Weatherfax 1.10.3
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.10.3+tarball

rgleason, 2024-12-23:
Steve please see weatherfax 1.10.4  or later which should be pushed to the catalog pretty soon.
If you still have problems please make another issue.
I have tested it for Windows and can do a capture. It isn't very clear, but that is probably due to it being a first for me. I am still learning.
Thanks for the report, sorry I did not get to it until now.

See https://github.com/rgleason/weatherfax_pi/issues/139#issue-1881033427

closed 2024-12-23 by rgleason