rgleason/weatherfax_pi #79: No more WRF in WeatherFaxInternetRetrieval_PWx_Euro_Atl.xml

ozolli, 2022-05-08:
As the title says, the links to wrf (all regions) has not been updated since March 2020.
[WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip](https://github.com/rgleason/weatherfax_pi/files/8647047/WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip)

rgleason, 2022-05-08:
What is WRF? 
[WIFR]_PWx_Euro_Atl.xml seems to be working fine.

ozolli, 2022-05-08:
There is no download error but the files are old: https://www.passageweather.com/maps/northsea/wrf/006.png is dated 04 march 2020 for example.

ozolli, 2022-05-08:
> What is WRF?

https://www.mmm.ucar.edu/weather-research-and-forecasting-model

rgleason, 2022-05-08:
Passageweather Low bandwidth Page of consolidated zip file  links
https://www.passageweather.com/download.htm

It appears we do not have the right coding for  PWx_Euro_Alt
Thanks for pointing this out.

**Passageweather Surface Wind**   Example

https://www.passageweather.com/maps/natlantic/m_gfs.htm

[image]

**Surface Wind Forecast**

Sun May 8 2022 GFS 000 hour   06UTC
https://www.passageweather.com/maps/natlantic/wind/000.png

Sun May 8 2022 GFS 003 Hour Forecast 09UTC
https://www.passageweather.com/maps/natlantic/wind/003.png

Sun May 8 2022 GFS 006 Hour Forecast 12UTC
https://www.passageweather.com/maps/natlantic/wind/006.png

Sun May 8 2022 GFS 009 Hour Forecast 15UTC
https://www.passageweather.com/maps/natlantic/wind/009.png

Sun May 8 2022 GFS 012 Hour Forecast 18UTC
https://www.passageweather.com/maps/natlantic/wind/012.png

rgleason, 2022-05-08:
It seems to be working properly on Win10
[image]
[image]
[image]

Are you sure about it not working for your OS?

ozolli, 2022-05-08:
> Are you sure about it not working for your OS?

You are showing me results for the GFS model. I said the WRF model is not updated anymore but the GFS and COAMPS work okay.

rgleason, 2022-05-08:
Ok, perhaps I should just remove wrf?

rgleason, 2022-05-08:
I have tried a number of WRF Wind and they are still working, I did find some missing, and perhaps they have lengthened out the scheduling to 6hrs for the ones further in the future?

rgleason, 2022-05-09:
I must be thick or something. I finally realized that your [WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip](https://github.com/rgleason/weatherfax_pi/files/8647047/WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip)  was there due to changes, so I compared to the existing. You have removed all the WRF Wind files have been removed, because they have not been updated.

Thanks very much. Will put yours in the data directory now.

rgleason, 2022-05-09:
https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.9.30.6

ozolli, 2022-05-09:
> I must be thick or something. I finally realized that your [WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip](https://github.com/rgleason/weatherfax_pi/files/8647047/WeatherFaxInternetRetrieval_PWx_Euro_Atl.zip) was there due to changes

Not your fault; I should be less laconic.

I'm working on UK Met Office Colour surface pressure charts like this : 
https://data.consumer-digital.api.metoffice.gov.uk/v1/surface-pressure/colour/2022-05-09T0000/FSXX00T_00.gif
Just waiting until tomorrow morning to see if the Hour variable is set correctly in the xml file.

rgleason, 2022-05-09:
Nice charts! Thanks.

rgleason, 2022-05-09:
@ozolli  If OK now, you can close.

closed 2022-05-09 by ozolli