rgleason/weatherfax_pi #81: UK Met Office scattered charts in xml files

ozolli, 2022-05-09:
For now we have some charts in 2 different locations : 
WeatherchartsUK - UK
UK MET - Europe

And I am working on new colour charts in another (temporary) location : 
UK Met Office - Atlantic NE

All those charts cover more or less the same geographical location which I have called Atlantic NE, although proper English would probably be NE Atlantic. This reflects NOAA zone definition.

Shall we group those 3 sources in the same group ?

rgleason, 2022-05-09:
I sounds to me like we should do as you suggest!

ozolli, 2022-05-09:
Sounds logical.

ozolli, 2022-05-10:
I have grouped "WeatherchartsUK - UK" and "UK Met Office - NE Atlantic" (my work now finished) in the same server "UK Met Office" and 2 regions "NE Atlantic (b/w)" and "NE Atlantic (color)".
Works fine.
I believe that, unfortunately, "UK MET - Europe" is dead or out of order since 27APR2022 so shall be discarded. I could not find those 500HPa geopot elsewhere.

rgleason, 2022-05-10:
Sounds very good to me. Let's wait until Dave is done so we don't get commits scrambled. Thanks.

rgleason, 2022-05-10:
WFIR_Europe.xml  WeatherChartsUK  the first 4 "Previous" are dated Sept. need to be removed if no replacement

ozolli, 2022-05-10:
Yes done already.

ozolli, 2022-05-13:
Changes done. Updated file: 
[WeatherFaxInternetRetrieval_Europe.zip](https://github.com/rgleason/weatherfax_pi/files/8685597/WeatherFaxInternetRetrieval_Europe.zip)

closed 2022-05-13 by ozolli

rgleason, 2022-05-13:
@Ozolli    I am going to try to push a new version of WF tonight using Jon's new version of  Testplugin frontend.
Will collect all your updates.  ...You've done an Awesome Job!!