rgleason/weatherfax_pi #82: WFIR_miscl.xml  needs updating badly

rgleason, 2022-05-10:
Bermuda - needs georeference fixes and one chart does not load.
French Polynesia - updated.
Chile MET - the two charts are ok, does the URL website have more available?
Weather SA - South Africa  - does not work. Need new URL's?  Different way to download like Aus-BOM?
Fiji MET -South Pacific - looks totally dated.  Is there a new URL?  or should we just drop this as it may be covered by Aus-BOM?
Mauritius MET - Mauritius - one overlay is dated and the other does not work.  Is there a new URL?  or should we just drop this too?

ozolli, 2022-05-11:
Taking care of this. Bermuda done.

rgleason, 2022-05-11:
ozolli you are doing a tremendous job. Thank you. I hoped that another user would assist.

rgleason, 2022-05-12:
For Jap MET,  the png files seem to download, but cannot be opened, because they are corrupted somehow.  "Failed to load input file..."   Also trying to open the file with other image software fails.  Faststone Image  says "unknown format"

rgleason, 2022-05-12:
These images appear to have date strings. I do not know if this is the best way to access them yet.

https://www.jma.go.jp/bosai/weather_map/data/png/20220512083531_0_Z__C_010000_20220512060000_MET_CHT_JCIasas_JCP600x512_JRcolor_Tjmahp_image.png

Open image in new tab
https://www.jma.go.jp/bosai/weather_map/data/png/20220512083531_0_Z__C_010000_20220512060000_MET_CHT_JCIasas_JCP600x512_JRcolor_Tjmahp_image.png

https://www.jma.go.jp/bosai/weather_map/#lang=en

B&W
https://www.jma.go.jp/bosai/weather_map/data/png/20220512083501_0_Z__C_010000_20220512060000_MET_CHT_JCIasas_JCP600x512_JRjmahp_image.png

rgleason, 2022-05-12:
There are also 3 choices at the bottom of  https://www.jma.go.jp/bosai/weather_map/#lang=en
which are PDF's some are color pdf's    
Analysis Chart,  
https://www.data.jma.go.jp/fcd/yoho/data/wxchart/quick/ASAS_COLOR.pdf
24-Hr Forecast Chart
https://www.data.jma.go.jp/fcd/yoho/data/wxchart/quick/FSAS24_MONO_ASIA.pdf
48-Hr Forecast Chart
https://www.data.jma.go.jp/fcd/yoho/data/wxchart/quick/FSAS48_MONO_ASIA.pdf

rgleason, 2022-05-12:
Trial still not working
```
 <Server Name="Japan Met" Url="https://www.jma.go.jp/bosai/weather_map/data/png/">
    <Region Name="Japan">
      <Map Url="%y%m%d%H.png" Contents="Analysis Now" Area="1" Hour="6;6;3" />
      <Map Url="%y%m%d%H.png" Contents="Analysis -6" Area="1" Hour="0;6;3" />
      <Map Url="%y%m%d%H.png" Contents="Analysis -12" Area="1" Hour="-6;6;3" />
      <Map Url="%y%m%d%H.png" Contents="Analysis -18" Area="1" Hour="-12;6;3" />
      <Area Name="1" lat1="0N" lat2="70N" lon1="90E" lon2="180E" />
    </Region>
  </Server>
```
I think "%y%m%d%H is ok but we are then missing some things.

ozolli, 2022-05-12:
The full URL is now like so : 
`https://www.jma.go.jp/bosai/weather_map/data/png/20220512083531_0_Z__C_010000_20220512060000_MET_CHT_JCIasas_JCP600x512_JRcolor_Tjmahp_image.png`
But there is a issued date at the beginning which makes the URL impossible to predict, like for the Fiji and also Mauricius. :-(

ozolli, 2022-05-12:
> Weather Online - Asia - Charts are dated. Are there any good charts at this URL? We really need Asia weather!

Wetterzentrale.de could be ok : 
`https://www.wetterzentrale.de/de/topkarten.php?map=12&model=gfs&var=1&time=6&run=6&lid=OP&h=0&tr=6&mv=0`

rgleason, 2022-05-12:
I haven't left yet (caulking 3 tubs before rentors).  I wonder if there is a kind of wildcard for the issued date or if it is always issued at a certain date/time. It is really too bad.

rgleason, 2022-05-12:
I am going to try to find if there is an ftp website for  Japan and Singapore

rgleason, 2022-05-13:
Wrote Singapore MET

```
Centre for Climate Research Singapore
36 Kim Chuan Road,
Singapore 537054
Email: NEA_MSS_Engage@nea.gov.sg

Ladies and Gentlemen,

We are a group of worldwide sailors who program for and use OpenCPN which is an Open Source ChartPlotter Navigation Program that is very popular with cruisers and worldwide sailors.  One of the plugins for this program, [Weatherfax_pi   ](https://opencpn-manuals.github.io/main/weatherfax/index.html)has useful links to Worldwide MET Marine Surface Forecasts, which allows sailors to download via internet or if the chart is sent via HF RadioFacsimile be able to receive that way and to overlay the forecast on the charts which OpenCPN uses.  We have been updating the data xml files for weatherfax (every 6 months generally).

For many years we were able to provide sailors with internet links to Singapore's MET Marine Forecasts,  Surface Analysis and Winds, and they were quite useful, but now the naming of the files has made it very difficult to do that.  The problem is that the files are named with both the valid data and time and the data and time it was prepared.  The date and time of preparation is not predictable, so we cannot provide good links, and will have to remove Singapore MET from our "WeatherfaxInternetRetrieval.xml file and sailors will no longer be able to access forecasts in that way.

For example:    http://www.weather.gov.sg/weather-marine-surface-wind

Uses links like: 

24hr forecast valid 8am 13 May 2022  (Updated at 9:30 pm Thu 12 May) .  This part of the file name "20220512_213000" cannot be predicted or is unknowable.

http://www.weather.gov.sg/files/marine/surfacewinds/surfacewinds_24Hr_20220512_213000.gif

48 hr forecaat valid 8 AM 14 May 2022 (Updated at 9:30 pm Thu 12 May)    This part of the file name "20220512_213000" is unknowable or cannot be predicted.

http://www.weather.gov.sg/files/marine/surfacewinds/surfacewinds_48Hr_20220512_213000.gif

72 hr forecast valid 8 AM 15 May 2022   (Updated at 9:30 pm Thu 12 May)    This part of the file name "20220512_213000" is unknowable.

http://www.weather.gov.sg/files/marine/surfacewinds/surfacewinds_72Hr_20220512_213000.gif

We do have string date and time routines which allow us to create the date/time links provided they are routine, regular, and are known.

Is there some alternative way to create a legitimate link to your forecasts with an easily known URL? Perhaps you have an ftp website, or some way that we can provide a link to your typical data, without having to know the date and time of creation of the file?

Any suggestions would be most appreciated.

Sincerely,

Rick Gleason

OpenCPN Weatherfax_pi maintainer.
```

rgleason, 2022-05-13:
Maybe we just use PW  for Asia as you suggested.
I just checked PWx Asian Pacific and PWx PAcific and I believe it covers everything needed in this region, isn't it?

ozolli, 2022-05-13:
I believe too.
It cannot hurt having more than one source for each location but in the pacific that's not easy!

rgleason, 2022-05-13:
Excellent.

rgleason, 2022-05-15:
Have downloaded, extracted and copied the  WFIR_misc.xml to the data directory.

rgleason, 2022-05-15:
Thanks, have this in the data directory now.  Closing

closed 2022-05-15 by rgleason