rgleason/weatherfax_pi #86: WFIR_ (NOAA, NOAA_OPC, NAVY) testing

rgleason, 2022-05-10:
These appear to be OK too, but others should test and report any issues here.

ozolli, 2022-05-12:
There are new color versions of the charts. I'll have a look at it.
https://ocean.weather.gov/Atl_tab.php

ozolli, 2022-05-13:
All tested and corrected. A lot of remapping involved ;-)
Updated files: 
[WeatherFaxInternetRetrieval_NOAA.zip](https://github.com/rgleason/weatherfax_pi/files/8688380/WeatherFaxInternetRetrieval_NOAA.zip)
[WeatherFaxInternetRetrieval_NOAA_OPC.zip](https://github.com/rgleason/weatherfax_pi/files/8688382/WeatherFaxInternetRetrieval_NOAA_OPC.zip)
[CoordinateSets.zip](https://github.com/rgleason/weatherfax_pi/files/8688389/CoordinateSets.zip)
(The CoordinateSets.xml is updated for all xml files).

ozolli, 2022-05-13:
All tested and corrected. Involved a lot of remapping.
Here are the updated files: 
[WeatherFaxInternetRetrieval_NOAA.zip](https://github.com/rgleason/weatherfax_pi/files/8688954/WeatherFaxInternetRetrieval_NOAA.zip)
[WeatherFaxInternetRetrieval_NOAA_OPC.zip](https://github.com/rgleason/weatherfax_pi/files/8688957/WeatherFaxInternetRetrieval_NOAA_OPC.zip)
[CoordinateSets.zip](https://github.com/rgleason/weatherfax_pi/files/8688960/CoordinateSets.zip)
(The CoordinateSets.xml is updated for all the xml files I modified).

rgleason, 2022-05-13:
Wow, ozolli. Super. Will pull your files together and jon has a new TP version.

rgleason, 2022-05-15:
Ozolli,  I've downloaded, unzipped and overwrote these into weatherfax/data directory, including coordinateSets.xml
(I reviewed coodinateSets.xml first)

rgleason, 2022-05-15:
Thanks have these in the data directory. Closing.

closed 2022-05-15 by rgleason