rgleason/weatherfax_pi #92: NOAA GOES Sat images

ozolli, 2022-05-17:
I could not map all the images as on some we can see the Earth horizon.
The Tropical Pacific Ocean one is included although not perfectly mapped but still usable.
As usual, the CoordinateSets.xml is included: 
[WeatherFaxInternetRetrieval_NOAA_OPC.zip](https://github.com/rgleason/weatherfax_pi/files/8709841/WeatherFaxInternetRetrieval_NOAA_OPC.zip)
[CoordinateSets.zip](https://github.com/rgleason/weatherfax_pi/files/8709851/CoordinateSets.zip)

ozolli, 2022-05-17:
The total new charts (all dates) for this update is 2337. Not bad !

ozolli, 2022-05-17:
And THAT is so cool and useful: 
[image]

rgleason, 2022-05-18:
Thank you Ozolli  these are good.  I've added WFIR_NOAA_OPC and COORD and have incremented version to 1.9.31.0, tested locally and pushed to master.

rgleason, 2022-05-18:
I found that in order to get updated coordiates, I had to remove my old version of coordinateSets.xml that was located in the USER Writable directory: C:\ProgramData\opencpn\plugins\weatherfax\coordinateSets.xml

Do you have to do that too?

rgleason, 2022-05-18:
PS I have quickly checked most regions and one or two images and they are all working well. I notice that Arpege has some images that refuse to download because they are so big and there is a note.

Also I downloaded so many images at once that wxfax_pi was running out of memory and misbehaving. This is a known issue, but most don't reach it.

ozolli, 2022-05-18:
Oh that's weird. It worked for me the first time when trying it and now it doesn't work anymore.
This is because I tried to maximise the MappingMultiplier as the resulting mapped image is less pixelized.
Apparently -and sadly- I need to be more conservative on some of them...
Let me check this before pushing a beta release.

ozolli, 2022-05-18:
Well, the only one that fails is the Arpege Europe map. I must have changed the value in CoordinateSets.xml manually by mistake as is it set to 1 for the other same maps (ECMWF, UK Met, DWD and GFS).
Have you seen it elsewhere ?
You can correct it on line 222 : MappingMultiplier="1.0000" (instead of 2.0000).

rgleason, 2022-05-18:
Goes images are good. Closing

closed 2022-05-18 by rgleason