rgleason/weatherfax_pi #100: Ubuntu 22.04 jammy

ozolli, 2022-05-20:
There are no plugins for the new LTS ubuntu 22.04-jammy.
This issue is not specific to WeatherFax though...

ozolli, 2022-06-13:
@leamas I uninstalled OCPN 5.6.2 ans the ppa repository, installed opencpn_5.6.0+dfsg1-1_amd64.deb and went to the plugin catalog settings.
The active setting is ubuntu-gtk3-x86_64:22.04 and there is no plugin available.
If I choose ubuntu-gtk3-x86_64:20.04 there are 42 available plugins and they work fine.
Did I miss something ?

leamas, 2022-06-13:
> Did I miss something ?

Yes. You missed that by using the plugins Settings menu you are outside of what's supported. Any bug report from users using this without instructions from devs is basically void. 

Facts in case is that with default settings the official 22.04 version from Ubuntu repos has full plugin support, the PPA version has not (yet). With other settings, well, this is just not supported.

ozolli, 2022-06-13:
> Yes. You missed that by using the plugins Settings menu you are outside of what's supported. Any bug report from users using this without instructions from devs is basically void.

Well, I try my best to help updating the weatherfax plugin and doing so have been asked by Rick to enable the beta catalog.
I haven't reversed the setting and saw the flavour missing when I updated unbuntu to 22.04, so this report.
I has no idea disabling CatalogExpert would enable the plugins, nor that the ppa version does not enable the plugins either...
Sorry to waste you time.

ozolli, 2022-06-13:
@leamas thanks for the explanations.
I know understand why I couldn't import a locally compiled plugin ;-)

rgleason, 2022-06-13:
@Ozolli  you wrote:  "no idea disabling CatalogExpert would enable the plugins"
NOTE: Actually that setting enables access to other PIM catalogs other than the "master" catalog.
It does not restrict access to any catalogs.

ozolli, 2022-06-13:
@rgleason I know this, and I now understand the official 22.04 version sets its CompatOsVersion to 20.04 like I did manually.
If it's set to 22.04 you get no plugins and this is why I raised this issue.
Closing now.

closed 2022-06-13 by ozolli