rgleason/weatherfax_pi #107: [Linux native & Flatpak] Selected entries displayed white

ozolli, 2022-05-23:
In the "Weather Fax" window, "Fax" menu list and in the "Internet Retrieval" window, "Servers" and "Regions" menu lists, the selected entries are displayed in white. In the Flatpak version they are still readable as "embossed" black, but they are invisible in linux native.

See this capture on Ubuntu: 
[image]

ozolli, 2022-05-23:
Flatpak capture: 
[image]

rgleason, 2022-05-23:
Thank you for the screenshots, this should help with understanding. We had an earlier report about this. 
I wonder if it is due to something missing in the xml?  or is it related to the OS and wx plugin programming used?

rgleason, 2023-09-05:
Is this still a problem in  Linux Native and Flatpak??

This is a Windows screen grab of the list.

[image]

ozolli, 2023-09-05:
Hi Rick,

I don't use native linux anymore so I can't tell you.
Flatpak is ok.

Olivier Zolli

rgleason, 2023-09-05:
Thanks, then I think we can close this now, as no other reports. We can reopen if necessary.

closed 2023-09-05 by rgleason