rgleason/weatherfax_pi #112: Bug: MacOS Menu

rgleason, 2022-05-28:
For MacOS we need to
        - simple fix: when activating weatherfax, not only open the Weatherfax window, but also make it the "active window" thus change the menu ...Against the rules (to change the menu), but it will [work](https://www.cruisersforum.com/forums/tags/work.html) and should be apparent to the user
        - [legal](https://www.cruisersforum.com/forums/tags/legal.html) fix, if there is need for context dependent menu entries, then "add an extra menu entry (e.g. weatherfax) to the existing menu bar" , but leaving the other entries in the standard [OpenCPN](https://www.cruisersforum.com/forums/tags/opencpn.html) menu there. In that new menu entry, we could find the retrieve points, ...

See Post 1512  https://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post3630585
or 
Post #1508
https://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post3626437

rgleason, 2025-01-12:
Manfred has this been resolved or not?

ManfredRad, 2025-01-15:
Yes, the above simple fix is implemented. Thanks.

closed 2025-01-15 by rgleason