rgleason/weatherfax_pi #115: Retrieve > Update Data Sources  confirmatory messages

rgleason, 2022-05-30:
In weatherfax.cpp at LINE 828

```
void WeatherFax::OnUpdateData( wxCommandEvent& event )
{
    if( DownloadFile( _T("WeatherFaxInternetRetrieval_NOAA.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_NOAA_OPC.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_NAVY.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_PWx_Amer_Atl.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_Aviation_Weather.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_PWx_Euro_Atl.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_LaMMA.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_Europe.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_PWx_India.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_Australia.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_PWx_Pacific.xml") ) &&
      DownloadFile( _T("WeatherFaxInternetRetrieval_Misc.xml") ) &&
      DownloadFile( _T("CoordinateSets.xml") ) ) {
        m_InternetRetrievalDialog.Load(true);
    }
    if( DownloadFile( _T("WeatherFaxSchedules.xml") ) ) {
        m_SchedulesDialog.Load(true);
    }
}
```
When these are all successfully download we would like a confirmation message:

```
"WeatherFaxInternetRetrieval_NOAA.xml"
"WeatherFaxInternetRetrieval_NOAA_OPC.xml"
"WeatherFaxInternetRetrieval_NAVY.xml"
"WeatherFaxInternetRetrieval_PWx_Amer_Atl.xml"
"WeatherFaxInternetRetrieval_Aviation_Weather.xml"
"WeatherFaxInternetRetrieval_PWx_Euro_Atl.xml"
"WeatherFaxInternetRetrieval_LaMMA.xml"
"WeatherFaxInternetRetrieval_Europe.xml"
"WeatherFaxInternetRetrieval_PWx_India.xml
"WeatherFaxInternetRetrieval_Australia.xml"
 "WeatherFaxInternetRetrieval_PWx_Pacific.xml"
"WeatherFaxInternetRetrieval_Misc.xml"
 "CoordinateSets.xml"
 "WeatherFaxSchedules.xml"
```
  
Downloaded successfully.

OR  List exactly which files did not download successfully.

Additionally we would like the User (writable)  CoordinateSets.xml file to be updated after this prompt is answered by the user.
`Update the User (writable) CoordinatedSets.xml?  Yes? No?
`
This prompt should also occur whenever the plugin is Updated or Installed.

rgleason, 2022-05-31:
or maybe a better prompt:

WeatherfaxInternetRetrieval files have been updated and URL's updated, we now need to remove the User (writable) CoordinateSets.xml (which may contain user customizations), so that the new file will work properly.  Yes? No?

ozolli, 2022-05-31:
Maybe move user's CoordinateSets.xml to CoordinateSets.xml.save and tell him he can merge his cutomizations if needed.

rgleason, 2022-05-31:
Yes, that's a good answer.

"Update Data Sources"  should:

1. Download and over write the xml data files in the Installation Directory (refer to the [Development Manual Installation Paths](https://opencpn-manuals.github.io/main/plugin-installer/Installation-paths.html) :
- WeatherfaxInternetRetrieval_(regions).xml 
- WeatherfaxSchedules.xml
- CoordinateSets.xml

2. Rename the 

- Windows 10  - User Writable and Actively Used Data Location: C:\ProgramData\opencpn\plugins\weatherfax\coordinateSets.xml
- Ubuntu      User Writable and Actively Used Data Location: $HOME/.var/app/org.opencpn.OpenCPN/config/opencpn/plugins/coordinateSets.xml

To coordinateSets.save.xml  (also rename user_coordinates.xml if it exists, to user_coordinates.save.xml)

rgleason, 2022-06-04:
OK, Ozolli. thanks. Changed to Flatpak in the post above.

rgleason, 2022-06-22:
Dave's PR for fixing this.
https://github.com/rgleason/weatherfax_pi/commit/ee64697df1d99ce0ff2f41232982ddac73477629

closed 2022-06-22 by rgleason