rgleason/weatherfax_pi #117: master-sd branch: rebase needed

leamas, 2022-05-30:
The whole rgleason/weatherfax_pi repo is  basically downstream from seandepagnier/weatherfax_pi. Since Sean seems to accept PRs, it's important to keep branches rebased to upstream before adding more stuff and diverging still more.

For that reason I suggest that the master-sd branch, the one I care about, is rebased on top of Sean's. That is, if you not already have the upstream remote defined, do it like

    $ git remote add upstream https://github.com/seandepagnier/weatherfax_pi

And then rebase the master-sd branch:

    $ git checkout master-sd 
    $ git remote update upstream
    $ git rebase upstream/master
    $ git push -f origin master-sd:master-sd

There are some simple conflicts which should not be hard to handle. 

Once master-sd is rebased, I can rebase my own changes and make a PR. Note that rebasing a branch we are cooperating on like master-sd should not be taken lightly. Still, it's the right thing to do in the current situation.

rgleason, 2022-05-31:
Alec, my preference is to stick with our current configuration and work on that. Thank you

leamas, 2022-05-31:
That leaves you and Sean with three options, no more and no less:
  - You continue to maintain a branch independent of Sean i. e., to create a new independent project also called a fork. This works fine, but you will have difficulties integrating any changes Sean does if you want to. Overall, maintaining a fork is always more work than integrating your changes with upstream.
  - You push the work to integrate "your" project  with Sean's upstream to Sean. This will cause him more work than if you rebase, since he is not aware of the changes you have done, and possible conflicts will be related to those. Most project maintainers refuses to do this, they require the PRs to be rebased to current master. The OpenCPN  project is an obvious example, as is every other I have worked in (there are some).
  - You rebase the master-sd branch onto Sean's master later, before making a PR against Sean. Certainly possible, but will be more complicated due to more conflicts

But, in the end you are the maintainer and it's your choice. I will make a PR against current master-sd. You have here three options:
1. Reject it
2. Merge it.
3. Rebase master-sd first. In this case I will rebase the PR, making everything compatible with Sean's upstream.

rgleason, 2022-05-31:
Jeez what a decision to have to make!

leamas, 2022-05-31:
Well, the fourth option is to ignore the PR. Otherwise, it's just how things are: you and Sean have diverged, and this must be handled somehow.

 I'd go for option 3) which is what I described in the first post.

rgleason, 2022-05-31:
Alec, I think I am going to merge your PR, you've made the options and situation clear to me. Thanks very much.

leamas, 2022-06-01:
Right. This means that you are now maintaining a fork. 

The flip side is that you are disconnected from Sean. You don't need to (or even can) make PRs against him, and you don't have to care about possible changes he makes.

The flop side is that you are disconnected form Sean. You are on your own, have to handle user and other issues yourself, and will have have problems integrating changes Sean does, if any. 

Goodbye, Sean.

rgleason, 2023-09-05:
done

closed 2023-09-05 by rgleason