rgleason/weatherfax_pi #122: Bugs: DWD weatherfaxInternetRetrieval_Europe.xml

rgleason, 2022-06-07:
@ozolli

There seems to be a problem with the 500mb   DWD > Europe files. 
They download but do not overlay. 

Attached are recently updated files to fix the  weatherfaxSchedule.xml issue regarding duration.
Clean up and trim white spaces.
Try and fix some issues with downloading DWD files that users noticed.
The data files are attached.
[data.zip](https://github.com/rgleason/weatherfax_pi/files/8854758/data.zip)

@ozolli  How do I get access to the DWD files, when I put in a url it shows forbidden, should I be using an FTP client?  I wanted to check to see the actuall 500mb files.

rgleason, 2022-06-07:
Having difficulty using Filezilla I guess it does not accept ftp
When I use https  the charts show up.
   https://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_ana.png  
and  https://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_024.png
and https://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_048.png
etc.

And they download ok I find these files in  C:\ProgramData\opencpn\plugins\weatherfax
They just are not overlaying properly. The screen jumps and nothing happens.

[image]

  Is  Area C possibly wrong?
      <Area Name="C" lat1="35N" lat2="80N" lon1="44E" lon2="44W" />

ozolli, 2022-06-08:
Ok, I'll look into it.

rgleason, 2022-06-08:
ozolli  Thank you!  I will do that. I also need to change coordinateSets, as Manfred suggested here

Please see  https://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post3636159

Any suggestions?

rgleason, 2022-06-08:
Available in  CS  1.9.34.1  tarballs
https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.9.34.1%20tarball

Also have pushed it to plugins master catalog and Beta catalog. Have not changed the coords.

rgleason, 2022-06-08:
I just tested downloading & overlay of all  DWD > Europe and after download and overlay of 7 the rest download, but do not overlay yet and image wizard does not appear either.

ozolli, 2022-06-09:
> ozolli Thank you! I will do that. I also need to change coordinateSets, as Manfred suggested here
> 
> Please see https://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post3636159
> 
> Any suggestions? I don't see "European" being used as he suggests.

He uses an old, non updated file.

ozolli, 2022-06-09:
> Available in CS 1.9.34.1 tarballs https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.9.34.1%20tarball
> 
> Also have pushed it to plugins master catalog and Beta catalog. Have not changed the coords.

Ah no ! It was just to test if it runs better with a smaller mapping size.
When you reduce the MappingMultiplier variable you get a smaller ram footprint.
But when you change the MappingMultiplier variable you also have to change other mapping variables for an accurate mapping. Here the chart is mapped over Greenland.
You should revert this commit.

rgleason, 2022-06-09:
Ok. Does it run better for rpi?  I see the mapping isn't right.  Also in Image Wizard the images are too small to see the Lat/Long so they should be enlarged anyway.

ozolli, 2022-06-10:
As I wrote ealier, it works fine on my PC with ubuntu and ok but slow on my RPi4.
I'm sure it's due to cpu time and ram needed to map the faxes.

The MappingMultiplier variable has a goal: The larger the number and the finer the rendered fax is. A small number means pixellated rendering but smaller ram occupation.

As I remapped nearly everything I tried to enlarge the MappingMultiplier each time to get beautiful rendering but I didn't think it would be an issue on several modest devices. It's mainly my fault as my computer is rather powerful.

I am keen to make a new "low powered" CoordinateSets.xml and keep the actual one for devices that can support it.
That would be great if there was an option to choose.

What do you think Rick ?

rgleason, 2022-06-10:
@ozolli 
I think it would be quite easy to manually support two coordinateSets.xml resolutions.
The low res version would be the default.
The user can then chose to copy over the low res version.
I would suggest distributing 

- CoordinateSets.xml   (default is low res)
- CoordinateSets-low.xml
- CoordinateSets-high.xml

Then to get high-res a user has to copy CoordinateSets-high.xml over CoordinateSets.xml
Later we can enhance this feature and add a control in Preferences.

ozolli, 2022-06-10:
Yes I agree.

rgleason, 2022-06-16:
@ozolli  We are going to be rebuilding all the TP plugins for additional OS. If you can send or attach the two coordinateSets.xml files to be included, that would be very helpful.  (I have about 20 plugins to do, but this is going to be done quite quickly, I hope, so I will need them in the next couple of days. Thanks so much.)

-     CoordinateSets.xml (default is normal res)
-     CoordinateSets-normal.xml
-     CoordinateSets-high.xml
-

rgleason, 2022-06-16:
Thanks so much, 4-5 days would be ok to. Leaving on a cruise next Wed.

ozolli, 2022-06-18:
Here we are : 
[CoordinateSets-normal.zip](https://github.com/rgleason/weatherfax_pi/files/8933824/CoordinateSets-normal.zip)
[CoordinateSets-high.zip](https://github.com/rgleason/weatherfax_pi/files/8933825/CoordinateSets-high.zip)
Hi hope there is no mistake...

rgleason, 2022-06-18:
Me too. Thanks so much! Will push to prod and plugins Master Catalog!
I could not or did not have the time to figure out how you get the coordinates with these.

rgleason, 2022-06-18:
I am going to close this issue now.

closed 2022-06-18 by rgleason