rgleason/weatherfax_pi #127: Bug:  Many XML regions are not shown

rgleason, 2022-06-22:
This bug was noticed in 1.9.37 while testing the new "Retrieve > Update Data Sources" which works well.
This bug has been around for awhile, and started before version 1.9.31  
I had not noticed it and have not traced it back further than that yet.

For some reason  NOAA only shows "Boston"  and none of the other regions, Pt. Reyes, New Orleans, Kodiak, Honolulu
NOAA-OPC similarly is not complete.
Every single Server is coming up short on Regions, missing many many weatherfaxes.

Why?

rgleason, 2022-06-22:
Now I don't think the "/" is the problem, I went back to look at the xml in 1.9.30.

ozolli, 2022-06-22:
Well, I cannot really help here as it works fine for me....

rgleason, 2022-06-22:
You have MacOS?

ozolli, 2022-06-22:
No, Linux Ubuntu or RPi4 or Windows 10.

rgleason, 2022-06-22:
Are you sure it is working "fine".   Does NOAA show all the regions?
Boston,  Pt. Reyes, New Orleans, Kodiak, Honolulu

ozolli, 2022-06-22:
Yes, all the regions, and I had to remap all of them for the low res version.

rgleason, 2022-06-22:
Which coorindateSets.xml are you using?

ozolli, 2022-06-22:
Hi res

ozolli, 2022-06-22:
Same with normal res.
You can't download the file or you can't display it ?

rgleason, 2022-06-22:
I am using Weatherfax 1.9.37.0 as installed by PIM.  I have checked both install dir and user dir and the coordinateSets.xml file are the same. I don't think that is the problem.  Also  Retrieve > Update Data Files works fine.

Not all Regions are displaying.. many of them.  We are also missing India. The files exist. The xml files show these regions.
It is bizarre.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rgleason, 2022-06-22:
This is frustrating. I thought we were done with this plugin.

ozolli, 2022-06-22:
I see that you miss some regions in the Aviation Weather server.
WFIR_Aviation_Weather.xml and the respective lines in CoordinatesSets.xml haven't changed for at least 13 months.
If you revert to, say, v1.9.20.0, do you still have this bug ?

ozolli, 2022-06-22:
I found a bug in 1.9.37 that is not present in 1.9.36 and that can be related to what you see:
In 1.9.36 when I click on a server which has only one region, i.e. French Polynesia or Chile, then when I click on the region I see all the charts.
In 1.9.37 I see nothing when clicking on the region.
However if I click the "All" button under the Regions list then when I browse to the region I see the charts.

rgleason, 2022-06-22:
It appears to me that WIFR_Aviation_Weather.xml is the only one that really works with all five of the Regions showing!
Infrared, Infrared color, Infrared-NWS-color, Visible and Water-vapor.
[image]

Just saw your post above....  Yes, now you've found the issue.

ozolli, 2022-06-22:
Let me check some things and report again.

rgleason, 2022-06-22:
Either the program is not reading the xml files properly, or the xml files have some very small issue. I think.

ozolli, 2022-06-22:
Well, I'm not sure now.
I wiped all WF data, reinstalled 1.9.36, updated data sources and it works fine.
The bug is in 1.9.37 but I don't believe in the data directory.

rgleason, 2022-06-22:
I find that 1.9.36 has the same issues after removing 
C:\ProgramData\opencpn\plugins\weatherfax
C:\ProgramData\opencpn\plugins\cache  ( removing weatherfax metadata)
%localappdata%\opencpn\plugins\weatherfax_pi  OR   C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi
removing the weatherfax_pi.dll from C:\Users\fcgle\AppData\Local\opencpn\plugins
and installing v1.9.36

rgleason, 2022-06-22:
In 1.9.36   Click NOAA, and Boston comes up in Regions  (missing 5 other regions), then click "ALL"
Just Boston shows, plus a lot of other regions, but not Pt Reyes, New Orleans, Honolulu, or Kodiak Alaska!
[image]

rgleason, 2022-06-22:
If you use Retrieve > Update data files  you will be updating to the current version 1.9.37 data.
We could go back to an earlier version and try to use the current data and see if it still has a problem.

rgleason, 2022-06-22:
I removed the weatherfax directories, data and using Import tarball installed 1.9.32.5  and tried it. I has the save issues with a clean installation.  Maybe you can try?
[weatherfax_pi-1.9.32.5-msvc-x86_64-10.0.14393-MSVC.tar.gz](https://github.com/rgleason/weatherfax_pi/files/8960109/weatherfax_pi-1.9.32.5-msvc-x86_64-10.0.14393-MSVC.tar.gz)

rgleason, 2022-06-22:
Now I tried updating the Master Catalog and downloading and installing 1.9.37 and the enabling it.
Now it is working!  Bizarro.

[image]

ozolli, 2022-06-22:
Tomorrow I will test again 1.9.36 vs 1.9.37 because I am not so sure the NOAA slow servers are the only issues here.

ozolli, 2022-06-23:
@rgleason @bdbcat 

I tested again 1.9.36 vs 1.9.37.
I made the screenshots from Windows because the native linux version of OpenCPN has the invisible selection bug (https://github.com/rgleason/weatherfax_pi/issues/107).

v1.9.37. When opening the intenet download window we get this:
[image]

All the servers and all the regions are already selected as if we had clicked the "All" servers and "All" regions buttons.

Then I clicked on "French Polynesia" server:
[image]

The "South Pacific" region is selected and we can see all the available charts.

Now click on "Chile" server:
[image]

The Chile region is selected but we cannot see the available charts.

Finally click on "French Polynesia" server again:
[image]

The "South Pacific" region is selected but this time we cannot see all the available charts.

This behaviour does not exist on v1.9.36. When opening the internet download window ony one server and one region is selected and the available charts are seen:
[image]

rgleason, 2022-06-23:
Do you have any thoughts on why this is happening? I have looked through the code, and don't see that any of the recent would cause this.  The only thing I can surmise is that we have multiple WFIR_<region> files that are read and displayed as if they are one XML file and there is some glitch in the coding for this.    Is that plausible?

rgleason, 2022-06-23:
Ozolli, actually  this is the commit list
https://github.com/rgleason/weatherfax_pi/commits/master
1. [1.9.36.0 update coordinateSets normal or high](https://github.com/rgleason/weatherfax_pi/commit/5dc47f500391cb1863909c500bde0e6897d5802e)
2. Then Dave's improvements [Correct XML data update file destinations](https://github.com/rgleason/weatherfax_pi/commit/ee64697df1d99ce0ff2f41232982ddac73477629)
3. Merge of those improvements [Merge branch 'rgleason:master' into master](https://github.com/rgleason/weatherfax_pi/commit/55e5c2144e0fea18296477f9d86f194115d039c8)
4. My increment of the version so that we can get the files into cloudsmith "prod" repository [1.9.37.0 bdbcat update data fix](https://github.com/rgleason/weatherfax_pi/commit/d92eab89c03888304db4aa877da2023998b8d024)

ozolli, 2022-06-23:
That's just an intuition, maybe this is here in src/InternetRetrievalDialog.cpp:

835          -  m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);
        850  + m_lServers->SetSelection(m_lServers->Append(it->Name));

(same for 858/873)

rgleason, 2022-06-23:
Thanks. Perhaps Dave will have a chance to look at this.

rgleason, 2022-06-24:
Thats great. You solved it. I will make that change

rgleason, 2022-06-25:
@ozolli 

```
I compiled the plugin with
m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);
and it works fine now.
```

Is this change made on LINE 850?  In lieu of

`850 + m_lServers->SetSelection(m_lServers->Append(it->Name));`

BTW I do not find this at line 835  or anywhere near line 835  
`835 - m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);`

I find with a search  "m_lServers->SetSelection"   that line 187, 515 and 850 have that search, but only 850 has the full line

Can you explain further please?  Thanks.

ozolli, 2022-06-25:
Rick,
The other one is on line 873.

rgleason, 2022-06-25:
yes
`873                m_lRegions->SetSelection(m_lRegions->Append(it->Name));`

is this line supposed to be 

`873        m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);`

???

ozolli, 2022-06-25:
No you keep m-lRegions. Just add: , it->Selected

rgleason, 2022-06-25:
OK ozolli, some confusion on my part.
I have made both Line 873 and 850 the same

`m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected); `

and built and push it up.  Please be explicit so this can be done efficiently. I don't like having a bunch of garbage in the commit history.  For example, like this???   Many variations
```
Line 850     m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);
Line 873     m_lRegions->SetSelection(m_lRegions->Append(it->Name, it->Selected); 
```

rgleason, 2022-06-25:
Sorry I am leaving to go sail. Will fix this in a couple of days.

ozolli, 2022-06-25:
Sorry Rick, I should have been more explicit.
This is correct:

> Line 850     m_lServers->SetSelection(m_lServers->Append(it->Name), it->Selected);
> Line 873     m_lRegions->SetSelection(m_lRegions->Append(it->Name, it->Selected);

rgleason, 2022-06-26:
This seems to work, sort of, changing just the first line 850, but "All" "None" commands don't work right and single regions do not display.
[image]

But when  I add line 873 it does not build locally and I get errors.

[image]

```
 aes.cpp
C:\Users\fcgle\source\weatherfax_pi\src\InternetRetrievalDialog.cpp(873): error C2664: 'int wxItemContainer::Append(const std::vector<wxString,std::allocator<_Ty>> &)': cannot convert argument 2 from 'bool' to 'void *' [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
          with
          [
              _Ty=wxString
          ]
  C:\Users\fcgle\source\weatherfax_pi\src\InternetRetrievalDialog.cpp(873): note: Conversion from integral type to pointer type requires reinterpret_cast, C-style cast or function-style cast
C:\Users\fcgle\source\weatherfax_pi\src\InternetRetrievalDialog.cpp(873): error C2143: syntax error: missing ')' before ';' [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
  af_vfs.cpp
  compression.cpp
```

rgleason, 2022-06-26:
So what are the lines used in your working version please?

rgleason, 2022-06-26:
I added the necessary ")" at the end before the ";" and I still get this error.

```
 Setup.cpp
  Track.cpp
C:\Users\fcgle\source\weatherfax_pi\src\InternetRetrievalDialog.cpp(873): error C2664: 'int wxItemContainer::Append(const std::vector<wxString,std::allocator<_Ty>> &)': cannot convert argument 2 from 'bool' to 'void *' [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
          with
          [
              _Ty=wxString
          ]
  C:\Users\fcgle\source\weatherfax_pi\src\InternetRetrievalDialog.cpp(873): note: Conversion from integral type to pointer type requires reinterpret_cast, C-style cast or function-style cast
  UUID.cpp
  WAVE.cpp
  aes.cpp
  af_vfs.cpp
  compression.cpp
  data.cpp
  debug.cpp
  format.cpp
  openclose.cpp
C:\Users\fcgle\source\weatherfax_pi\src\libaudiofile\WAVE.cpp(1135): warning C4267: '=': conversion from 'size_t' to 'uint16_t', possible loss of data [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
C:\Users\fcgle\source\weatherfax_pi\src\libaudiofile\WAVE.cpp(1142): warning C4267: '=': conversion from 'size_t' to 'uint16_t', possible loss of data [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
C:\Users\fcgle\source\weatherfax_pi\src\libaudiofile\WAVE.cpp(1182): warning C4267: 'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
C:\Users\fcgle\source\weatherfax_pi\src\libaudiofile\WAVE.cpp(1189): warning C4267: 'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data [C:\Users\fcgle\source\weatherfax_pi\build\weatherfax_pi.vcxproj]
  PacketTable.cpp
  query.cpp
```

rgleason, 2022-06-26:
I am going to remove weatherfax from the catalogs.

rgleason, 2022-06-26:
Also I notice there is a loose "} in this procedure and the { and } aren't paired properly.

rgleason, 2022-06-26:
I was wrong, the { and } do match up.   It it the conversion from integral to pointer type...