rgleason/weatherfax_pi #131: [Tropical cyclone danger] Wrong mapping

ozolli, 2022-09-24:
Hello Rick,
I was looking at Fiona hurricane and noticed Tropical cyclone danger chart is not mapped correctly.
It is available in NOAA Boston and New Orleans regions.
I have a new computer and still not installed git so here is the archive with the updated files:
[NOAA.zip](https://github.com/rgleason/weatherfax_pi/files/9639523/NOAA.zip)
Cheers,
Olivier.

rgleason, 2022-09-24:
Awesome Olivier.  Thank you.  I will unzip and PR a new version.

closed 2022-09-25 by ozolli

rgleason, 2022-09-25:
Not done, yet.

ozolli, 2022-09-25:
Yes, but as you said you will take care of it I considered it closed ;-)

rgleason, 2022-10-05:
ozolli 
I finally got this done, haven't pushed it yet because I am updating the frontend too.
I found the high res was OK, but the normal coordiatesSets.xml was still offset.
It appears they widened the area of the overlay due to the wandering hurricanes now.

ozolli, 2022-10-05:
Hi Rick,
I don't understand ; These are exactly the same lines in normal and high as the mapping multiplier is the same at 1...
On my computer they both show correctly.

rgleason, 2022-10-05:
Well, something was amiss when I tried it on my Win machine.
So I just changed it. Will be pushing it later, but build-flatpak-arm64 refuses to build now.