rgleason/weatherfax_pi #133: New South Atlantic chart from Marinha Do Brasil

ozolli, 2022-11-19:
Hi Rick,

Found a much better one that the chart from Chile Met:
[image]

Here are the files:
[WFIR_Misc.zip](https://github.com/rgleason/weatherfax_pi/files/10047416/WFIR_Misc.zip)

Cheers,
Olivier.

ozolli, 2022-12-07:
Just found a wrong mapping on Aviation Weather - Sat Images Water-Vapor - Atlantic.
Corrected CoordinateSets files attached :
[CoordinateSets.zip](https://github.com/rgleason/weatherfax_pi/files/10179155/CoordinateSets.zip)

rgleason, 2022-12-08:
Oliver, thank you. I will be updating PI for 5.8.1 shortly and will make these changes then.

rgleason, 2022-12-25:
Olivier,  It's wonderful you found a good Surface chart for South America,
I am applying these now.  First the surface chart files, then the coord files. 
Wxfax is also updated for 5.7.1   Will push these shortly.  Thanks.

closed 2023-09-05 by rgleason