rgleason/weatherfax_pi #134: strange incompatibility with OpenCPN 5.8.2

filochard, 2023-05-14:
Hi Rick
I still use to build rpms for OpenCPN and its plugins
I built a rpm of weatherfax 1.9.54 for OpenCPN 5.6.2, installed it as an update and it's Ok

I'm building rpms now for OpenCPN 5.8.2 and test them before proposing them as updates for rpm based Distributions
(I had to patch OpenCPN 5.8.2 so that it can find the libraries installed by rpms instead of using the Plugin Manager)
It's OK for most of those plugins  except for weatherfax_pi that is said not to be compatible with OpenCPN 5.8.2...

Beside this I'm testing the flatpak version of OpenCPN 5.8.2 and its plugins downloaded with the Plugin Manager and there weatherfax_pi is installed, may be activated and works !

I can't see what part of the code is implied for this strange incompatibility

rgleason, 2023-05-14:
That is a stumper for me, but I suspect that it is related to where the user modified files are saved. Please provide more detail.

filochard, 2023-05-14:
Hi Rick
here is an extract of the log

15:33:25.286 MESSAGE plugin_loader.cpp:275 Checking plugin candidate: /usr/lib64/opencpn/libweatherfax_pi.so
15:33:25.286 MESSAGE plugin_loader.cpp:340 Checking plugin compatibility: /usr/lib64/opencpn/libweatherfax_pi.so
15:33:25.286 MESSAGE plugin_loader.cpp:1147 Plugin is compatible by elf library scan: true
15:33:25.286 MESSAGE plugin_loader.cpp:1210 PluginLoader: Loading PlugIn: /usr/lib64/opencpn/libweatherfax_pi.so
15:33:25.287   ERROR dlunix.cpp:120 Échec du chargement de la bibliothèque partagée « /usr/lib64/opencpn/libweatherfax_pi.so »: /usr/lib64/opencpn/libweatherfax_pi.so: undefined symbol: afSeekFrame
15:33:25.287 MESSAGE plugin_loader.cpp:1265    PluginLoader: Cannot load library: /usr/lib64/opencpn/libweatherfax_pi.so

filochard, 2023-05-14:
Hi again
Sorry for the noise : 
A correctly built rpm works !
I had applied an incorrect patch to use the audiofile-devel source from our distribution instead of the bundled one,
That was the reason why afSeekFrame was undefined (audiofile.h had not been found when compiling)

closed 2023-05-14 by filochard