rgleason/weatherfax_pi #138: test weatherfax on debian12 bookworm

hreuver0183, 2023-09-04:
Q&D test:
plugin not available from standard repository.

compiling:
'git clone https://github.com/rgleason/weatherfac_pi
'mkdir build
'cd build
'cmake ..
'time make -j 6 
'real	0m14.680s
'user	1m15.286s
'sys	0m5.748s

importing plugin with plugin manager import:
Error, import plugin tarball does not contain required metadata

Putting the binaries directly in .local yields a working weatherfax (I only tested the faxes via internet).
Seems useable, but the packaging needs some care.
At least no segfaults with wxgtk.

rgleason, 2023-09-04:
@hreuver0183 @bdbcat  We really need to make sure that these bookworm work. What OS are you using? x86_64 or armhf or arm64?  
I am wondering if the target is not right.

hreuver0183, 2023-09-04:
Te comment on 'putting directly in .local' is as follows:
Goto git/weatherfax_pi/build
mkdir tmp
cd tmp
tar xzf ../weatherfax_pi-1.9.65.0-debian-x86_64-12.tar.gz 
mv weatherfax_pi-1.9.65.0-debian-x86_64-12/usr/local/ .local
tar caf ../weatherfax_pi-1.9.65.0-debian-x86_64-12.txz .local/

This creates a  weatherfax_pi-1.9.65.0-debian-x86_64-12.txz which can be unpacked in the home-directory. Than you activate the plugin.
(I know, it is rough, but it is the easiest way to get the plugin working in Debian.)

hreuver0183, 2023-09-04:
I can test in arm64 (odroid) and amd64 (i7).

Arm64 would not be a supported platform (for the rpi4 there is raspberry pi os/bullseye, but no bookworm).
The Odroid N2+ would be best supported by Ubuntu 22.04. Flatpak is only an option if all needed plugins are available.
(I use the Odroid N2+ with Bookworm and that is not nearly mainstream.)

For Amd64 I'd really prefer native support with plugins. Currently I have the o-charts_pi and ocpn_draw_pi selfcompiled (o-charts is the only plugin where the tarball import works).

The tests above are x86_64 or amd64 (intel).

rgleason, 2023-09-04:
Thank you @hreuver0183  you've given us excellent information on the bug. We should be able to fix it. Please watch, for a link to test. Thanks again.

rgleason, 2023-09-04:
The key part of the metadata file is the "target"

weatherfax has this for bookworm x84_64

```
<target>debian-wx32-x86_64</target>
<build-target>debian</build-target>
<build-gtk></build-gtk>
<target-version>12</target-version>
<target-arch>x86_64</target-arch>
<tarball-url>
https://dl.cloudsmith.io/public/opencpn/weatherfax-prod/raw/names/weatherfax_pi-1.9.65.0-debian-x86_64-wx32-12-bookworm-tarball/versions/v1.9.65.0/weatherfax_pi-1.9.65.0-debian-x86_64-12-bookworm.tar.gz
</tarball-url>
```

Similar for the others

```
<target>debian-wx32-armhf</target>
<build-target>debian</build-target>
<build-gtk></build-gtk>
<target-version>12</target-version>
<target-arch>armhf</target-arch>
<tarball-url>
```

I believe that according to the [TP Template Table](https://github.com/jongough/testplugin_pi/issues/284)

**wx32 is assumed**   so the metadata target is NOT accepted

<html><body>

Debian 12 Bookworm |   | wx32 is assumed |  
-- | -- | -- | --
debian-arm64-12-bookworm | Jammy* | wx32 gtk3 | O58 ADD
debian-armhf-12-bookworm | Jammy* | wx32 gtk3 | O58 ADD
debian-x86_64-12-bookworm | Jammy* | wx32 gtk3 | O58 ADD

</body>
</html>

rgleason, 2023-09-04:
@bdbcat 

**PluginSetup.cmake**

```
elseif(UNIX)
    # Some linux dist:
    if("$ENV{BUILD_ENV}" STREQUAL "debian" OR "$ENV{BUILD_ENV}" STREQUAL "DEBIAN")
        set(PKG_TARGET "$ENV{BUILD_ENV}")
        if( NOT "$ENV{WX_VER}" STREQUAL "" )
            set(PKG_TARGET_WX_VER "-wx$ENV{WX_VER}")
        endif()
        message(STATUS "${CMLOC}PKG_TARGET_WX_VER: ${PKG_TARGET_WX_VER}, WX_VER: $ENV{WX_VER}")
    elseif(("$ENV{BUILD_ENV}" STREQUAL "ubuntu" OR "$ENV{BUILD_ENV}" STREQUAL "UBUNTU") AND
```

It looks to me like we just have to not have any value for "WX_VER"  in the circleci/config.yml???

```
    build-debian-x86_64-12-bookworm:
        machine:
            image: ubuntu-2004:202104-01
        environment:
        - OCPN_TARGET: bookworm
        - BUILD_GTK3: true
        - WX_VER: 32
        - DEPLOY_USE_ORB: true
        - BUILD_ENV=debian
        - DOCKER_IMAGE=debian:bookworm
        steps:
        - checkout
        - run: chmod a+x ci/*.sh
        - run:
            command: ci/circleci-build-ubuntu-docker.sh
            no_output_timeout: 30m
        - deploy-code

```

So  there would be no  "-WX_VER:32"   ????

rgleason, 2023-09-04:
This can't be right because this parameter determines which version of wxWidgets to use!
So we have to adjust somewhere else.

bdbcat, 2023-09-04:
I'll take a look. I am working in in WEFAX now, and this was on my list.
If wefax is based on TP, there might be problems elsewhere.  Or maybe wefax missed an update? Or?
Anyway, I will include this fix in my next PR, tonight.

rgleason, 2023-09-04:
@bdbcat I am afraid it runs through all TP plugins  TestPlugin needs to be fixed too.  Thanks.

rgleason, 2023-09-04:
I think Jon might have tried to fix it here https://github.com/jongough/testplugin_pi/commit/f4a4444cc912b17f6c9e9c878c8e284b1ddd7058  but not sure about that.

bdbcat, 2023-09-05:
Rick...
I have corrected WeatherFax_pi for debian-12/x86_84.
But, a new problem:  MacOS builds are failing due to some change in CircleCI which I am working to understand.
In fact, I think that all plugin builds for MacOS will fail now.  So this is clearly not due to something we broke in WeatherFax_pi.  You may like to try a build of testplugin_pi in CCI, from the current sources.  I expect it will fail.

So, I will hold my PR until this MacOS problem is understood, or self corrects.  After all, David Burch is mostly a mMac user...

rgleason, 2023-09-05:
Yes, tonight I have gotten several macOS notices for other plugins  Deviation, Testplugin and Weatherfax.
I believe we have the build correct and hope that this is temporary.  Thanks.

bdbcat, 2023-09-05:
As sometimes happens in the CI alternate universe, the MacOS issue magically fixed itself.  My builds of weatherFAX_pi are completing correctly now.
PR coming.
Dave

rgleason, 2023-09-05:
Ok, good news. Since you've fixed the wx32 in debian-12-bookworm builds, I am closing this.

closed 2023-09-05 by rgleason