rgleason/weatherfax_pi #144: RFE: adding graphic indexes to ASCAT part of WeatherFax_Pi

starpathnav, 2023-09-06:
Recommend adding the two graphic index images to the the top of the Regions list, which then has all individual regions presented alphabetically.
      Index Europe
      Index North America
      Adriatic
      Bermuda
      Biscay...
The KML files for the index images are attached as well as the georeferencing. With this the user finds the region they want as well as the valid times of the data and then can find the region in the list below. Once available to the public I will make a video that can be referenced for the use of the data.
[Weatherfax_pi ASCAT RFE.zip](https://github.com/rgleason/weatherfax_pi/files/12539663/Weatherfax_pi.ASCAT.RFE.zip)

rgleason, 2023-10-28:
> David...
> 
> Following your detailed description, and using your KML files, we have completed an initial update to weatherfax_pi incorporating the ASCAT data overlay function.
> 
> The Plugin is available for download and import now (for Windows10).
> 
> https://dl.cloudsmith.io/public/opencpn/weatherfax-beta/raw/names/weatherfax_pi-1.9.67.0-msvc-x86_64-wx32-10.0.20348-MSVC-tarball/versions/1.9.67.0+734.73b8994/weatherfax_pi-1.9.67.0-msvc-x86_64-wx32-10.0.20348-MSVC.tar.gz
> 
> We would appreciate your feedback before publishing the plugin update to a wider audience.
> Good Luck

rgleason, 2024-01-23:
This is Done, but ASCAT. but still needs work, handled in another issue..

rgleason, 2024-01-23:
https://github.com/rgleason/weatherfax_pi/issues/157

closed 2024-01-23 by rgleason