rgleason/weatherfax_pi #145: TP Streamline

rgleason, 2023-09-06:
LIBS folders  are unique to weatherfax,

- curl           <--- Is this really needed?
- glshim      <--- I don't think this is needed anymore
- plugingl    <---I think this is the same as pluginDC
- wxcurl       <--- Is this needed?
- zlib-1.2.3    <---Weather_routing needs zlib1.lib  maybe this would help

Can any of these be eliminated or used by other plugins?

rgleason, 2023-09-07:
**weatherfax/streamline   branch**
https://github.com/rgleason/weatherfax_pi/tree/streamline

Last output result building with Windows.
[output.txt](https://github.com/rgleason/weatherfax_pi/files/12543896/output.txt)

bdbcat, 2023-09-07:
Once again, on the libs folder:
We want the libs folder as checked-in in git for weatherfax_pi to be EXACTLY THE SAME as the "golden version" we are using, which is "libs" from watchdog_pi.  If weatherfax needs something additional in libs, we will back-add it to watchdog.
We do not care if watchdog does not use some elements of "libs".

I have posted a PR to correct builds, and some of above discussion.

rgleason, 2023-09-07:
Thankyou Dave.  -- Huge help.
I will double check "libs"  comparing weatherfax libs  to watchdog libs and backfill the missing libs.

I think I can merge your PR now.

Regarding the folder and file list above... 
I think what I have learned from your fix is that

****
What files are actually necessary and used?
Can any of these be eliminated or used by other plugins?

These LIBS folders are unique to weatherfax,

    libs/curl <--- Is this really needed?   FOR LINUX BUILDS
    libs/glshim <--- I don't think this is needed anymore   NOT NEEDED
    libs/plugingl <---I think this is the same as pluginDC    YES
    libs/wxcurl <--- Is this needed?    YES  Also used for ftp download capability for DATA
    libs/zlib-1.2.3 <---Weather_routing needs zlib1.lib maybe this would help?  <---Would this replace WR zlib1.lib?

Other folders and files

    src/libaudiofile <--- is for the audio portion, can it be moved to libs?  MAYBE LATER
    include/GL <--- is this necessary?       NO NOT REALLY COULD USE LIBS
    include/msvc/inttypes.h <--- is this necessary? can it be moved?    DONT KNOW
    include/ <--- many audio header files?    MAYBE LATER
    buildwin/portaudio_x86.dll       I GUESS WE JUST LEAVE THESE HERE?
    buildwin/portaudio_x86.lib
    buildwin/PVW32Con.exe
    buildwin/zlib.lib (used in Weather_Routing )

rgleason, 2023-09-07:
https://github.com/rgleason/weatherfax_pi/commit/9ca14f7d2379d083edcdfd720c68280b12412bc6

closed 2023-09-07 by rgleason

bdbcat, 2023-09-07:
Rick...
Do not ask the question "Is this needed" for anything in libs.  It might be needed for some other plugin.
Anyway, pretty soon libs will go away completely, in favor of opencpn-libs.

re: Files particular to weatherFAX_pi:

src/libaudiofile <--- is for the audio portion, can it be moved to libs?  MAYBE LATER
  This is unique to weatherfax, no-one else will need it.  Keep specific here.

include/GL <--- is this necessary?       NO NOT REALLY COULD USE LIBS
Delete it, and try the build.  General rule:  If a required header file is missing, the build will fail in a pretty obvious manner.

include/msvc/inttypes.h <--- is this necessary? can it be moved?    DONT KNOW
Delete, and try.

include/ <--- many audio header files?    MAYBE LATER
Specific to weatherFAX. Keep.

buildwin/portaudio_x86.dll       I GUESS WE JUST LEAVE THESE HERE?
buildwin/portaudio_x86.lib
buildwin/PVW32Con.exe
buildwin/zlib.lib (used in Weather_Routing )

Yes to all.