rgleason/weatherfax_pi #157: RFE: ASCAT Coordinates are needed

rgleason, 2023-10-28:
Display is not seamless and automatic.  Coordinates File is needed

[image]

ozolli, 2023-11-10:
Hi Rick,
Please find the attached zip file with coordinates, ascat xml file corrected (typos) and AviationWeather xml updated (url has changed).
Cheers,
Olivier.
[ASCAT-AviationWeather.zip](https://github.com/rgleason/weatherfax_pi/files/13318772/ASCAT-AviationWeather.zip)

rgleason, 2023-11-10:
Dear Olivier, Thank you so much! I will be able to make these changes in about 7 weeks.

ozolli, 2023-11-10:
My pleasure Rick!
I am not actively monitoring github so anytime you need me to do some coordinates or trying to fix an url or whatever send me an email or tag me.

ozolli, 2023-11-23:
I found an error in the ASCAT file so don't use it for now.
Bruce Balan wants to add more ASCAT locations, me too.
So, Rick, please wait for us before issuing a new version.
I will ask Bruce to send me his locations so I will post the updated files here.

rgleason, 2023-11-23:
OK. Also you and others report that aviation url has changed, so that needs updates too.

ozolli, 2023-11-23:
Already done in the 1st zip I posted here 2 weeks ago.

rgleason, 2024-01-22:
@ozolli    

I am going to be updating weatherfax over the next few days, should I include the file above?  Thanks.

> found an error in the ASCAT file so don't use it for now.
> Bruce Balan wants to add more ASCAT locations, me too.
> So, Rick, please wait for us before issuing a new version.
> I will ask Bruce to send me his locations so I will post the updated files here.

rgleason, 2024-01-23:
RFE: ASCAT now a part of WeatherFax_Pi - Update Manual #144

rgleason, 2024-01-23:
@ozolli   
I've finally added your https://github.com/rgleason/weatherfax_pi/files/13318772/ASCAT-AviationWeather.zip
to the data directory, tried it out and it seems to work, but there are some selections which do not seem to show anything....
or are looking for coordinates.
The new version is  version 1.9.111.0  found here when it completes

 https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.9.111.0

I will be pushing this up to PIM, but if you and Bruce can send me some Data updates I will certainly add them.

Thanks. so much!

ozolli, 2024-01-23:
@rgleason 
Bruce has sent me a lot of ASCAT entries.
I have to include them and post the new files here.
I'll have a look at AviationWeather.
Cheers,

ozolli, 2024-01-23:
@rgleason 
AviationWeather is all good on my side.

rgleason, 2024-01-23:
https://github.com/rgleason/weatherfax_pi/issues/156

filochard, 2024-01-25:
Hi
Some context first : 
I continue to maintain rpm packaged OpenCPN and legacy installed plugins for a Linux distribution (thanks to the devs who helped me to find a way to continue to do so, instead of using flatpak)

With a an other contributor, we  have tested the
 WeatherFaxInternetRetrieval_Aviation_Weather.xml from the zip (overwriting the previous one)
It works like a charm
Thanks to ozolli

We wait for the next WeatherFaxInternetRetrieval_EUMetSat_ASCAT.xml to test it too

rgleason, 2024-01-25:
@ozolli  Thanks!  We'll build & push when you guys finish.

> Bruce has sent me a lot of ASCAT entries.
> I have to include them and post the new files here.
> I'll have a look at AviationWeather. - [OK}  Thanks.
> Cheers,

filochard, 2024-02-03:
Hi Rick and ozolli
@ ozolli
Is there something new about the inclusion of ASCAT entries that Bruce sent to you
I already tested the Aviation_Weather xml file that you provided : it's OK  :o)
I will test the improved WeatherFaxInternetRetrieval_EUMetSat_ASCAT.xml when it's ready...

I intent to update our rpm of Weatherfax_pi but I will do this only when I can add Aviation_Weather **and** ASCAT xml files

filochard, 2024-02-13:
Hi Rick and ozolli
Bad news
AviationWeather xml is no more useable !!!
You may remove it from the weatherfax/data !
There is now only access to an interactive map and no more to cached data

I wrote this to mailto:aviationweather.webmail@noaa.gov  : 

  Category: Bug
    Subject: https://aviationweather-cprk.ncep.noaa.gov/ not reachable

    I have tried several times to go to this site https://aviationweather-cprk.ncep.noaa.gov/ to download satellite images

    I could do this in january 2024 but that seems no more possible
    has this address been changed ? or this site is temporarily not reachable ?

And here is the answer from
Lewis.Kanofsky@noaa.gov : 

Hi Philippe,

Please note that the URL for our website changed on October 16th as part of a major website upgrade. The new address is https://aviationweather.gov/ (note the lack of "www'). If you are using a link which contains "cprk" or "bldr", those links will no longer work. They have not been operationally supported since October 16th and the servers were fully decommissioned last week. Please update your links to use the new site. We have a URL migration guide available at https://aviationweather.gov/help/#url and a video tutorial for the new site at https://www.youtube.com/watch?v=Sdkx3UWeflI .

As part of the website upgrade, the static satellite images have been discontinued. This is because the new website is centered around an interactive and fully customized map, and the satellite imagery has been incorporated into that map. To view satellite imagery on the new site, click the "Weather" menu at the top of the screen, then click "Observations". After the map loads, click the layer icon in the upper right corner of the map. This is where you can turn on/off the satellite imagery and switch between different types such as visible, IR, and water vapor. Additional customization settings are available by clicking the gear icon to open the settings menu (press the blue “close” button to apply any changes).

Kind regards,
Lewis Kanofsky

rgleason, 2024-02-13:
Filochard, Thankyou for writing regarding Aviation Weather. 
This is a different Issue. and will require some response,  but I will respond here.
I notice that there are static images for example
https://aviationweather.gov/graphics/
https://aviationweather.gov/gfa/#winds
Also Faxes  https://aviationweather.gov/fax/
We will need to review what they offer and decide what if anything is appropriate to include.

filochard, 2024-02-14:
Hi Rick
I already had a look at : 
https://aviationweather.gov/graphics/
https://aviationweather.gov/gfa/#winds
indeed they mostly concern USA flight areas

The faxes from https://aviationweather.gov/fax/ 
are more useful : 
you may choose one area (11 areas are covered)
for each area you may choose the flight level (I think that more than 050 = 5000 feet is useless for sailors) 
and the hour

But forecast faxes are already provided (at sea level) by many other providers and I don't think aviationweather faxes are  really needed now

The satellite views that were provided by https://aviationweather-cprk.ncep.noaa.gov/ (the interesting part of this site for sailors)
are no more reachable 
They are now replaced by this : 
https://aviationweather.gov/gfa/#obs 
which is updated continuously and can be zoomed 
but that can't be downloaded...

rgleason, 2024-02-15:
Please post any further comments about Aviation Weather in the Issue above.

This Issue is about  ASCAT updates.
See https://github.com/rgleason/weatherfax_pi/issues/157#issuecomment-1905476772
Bruce and Ozzolli may have some updates to ASCAT.

filochard, 2024-02-15:
> Please post any further comments about Aviation Weather in the Issue above.
> 
> This Issue is about ASCAT updates. See [#157 (comment)](https://github.com/rgleason/weatherfax_pi/issues/157#issuecomment-1905476772) @ozolli and Bruce Balan may have some updates to ASCAT.

Hi Rick
Sorry about my out of topic comment
As I had seen that the same zip was about aviationweather and ASCAT in the second comment I thought it was the good place to talk about aviationweather

rgleason, 2024-02-15:
No problem. filochard. Just good to have them separate.

svMigration, 2024-04-17:
Hi Rick,
Olivier is swamped and I told him I would update the coordinatesets.xml files with all the ASCAT info for the Pacific. There are a couple of zips posted in this thread and I don't want to work on the wrong one. Can you point me to the correct one to work on?
Thanks,
Bruce

filochard, 2024-04-17:
> Also Bruce you can fix the Aviation problem by [removing those file](https://github.com/rgleason/weatherfax_pi/issues/166)s, or perhaps, if you have time you can use some of the images they have the Aviation website now. Filochard wrote them about the [change to an interactive map](https://github.com/rgleason/weatherfax_pi/issues/157#issuecomment-1941974341)
> 
> They have fax charts here https://aviationweather.gov/fax/ Maybe. https://aviationweather.gov/data/products/fax/F24_sigwx_hi_a.gif These look useful https://aviationweather.gov/fax/ Some of the Canadian coastal ones might be useful.
> 
> But this is a different job and geography. Sorry, I am busy with update of the User Manual these days. Hope that is a help.

Hi Rick
About AviationWeather  the faxes that they propose now have only an interest for aviation : they provide forecast at different flight levels (FL)  each FL number indicates hundred feet (that means FL050 = flight level 5000 feet  FL100 = flight level 10000 feet)
Only the lowest altitude (5000 feet) might have a little interest for sailors but it is nevertheless far from what we get at the sea level (the wind is usually less strong at the sea level than at 5000 feet, and its orientation may be different...)

rgleason, 2024-04-17:
What you say is true, but the lower level ones are like the 500mb faxes.

svMigration, 2024-04-19:
Thanks for all that, Rick. We're heading away from internet land for a bit so I'll start on this when I have time. Hope it's OK if I email you or post with questions.
Cheers,
B

rgleason, 2024-12-20:
I am going to remove Aviation Weather as you all suggest. 
Also the ASCAT indexes are almost working and we have a specific issue for that.
So this is being closed.

closed 2024-12-20 by rgleason