rgleason/weatherfax_pi #162: WeatherFaxInternetRetrieval_Aviation_Weather.xml no more useable

filochard, 2024-01-21:
The internet site https://aviationweather.gov  doesn't provide satellite img anymore
Nevertheless 
https://aviationweather.gov/fax/  provides lots of faxes under the sigWx tab

WeatherFaxInternetRetrieval_Aviation_Weather.xml may be rewritten for these new link

Nevertheless these are faxes for meteo at flight altitude !!!

filochard, 2024-01-21:
Maybe it would be more rational to suppress this provider that is not very useful for sailors

Ask Sean DePanier what he thinks about that

rgleason, 2024-01-22:
@filochard   Thanks, see https://github.com/rgleason/weatherfax_pi/issues/157
these URL corrections are in that file, but Ozolli  and Bruce Balan also want to update that file.

I realize aviation is for higher levels, however we also use 500mb files to determine how upper level winds are working to affect our lower level weather, so aviation files are a very good indicator for that and they are very graphical.

rgleason, 2024-01-22:
@filochard  This is going to be fixed with next update.  Can we close this now?

closed 2024-01-23 by rgleason