rgleason/weatherfax_pi #165: Update portaudio?  How to update these lib files?

rgleason, 2024-01-23:
The portaudio files are ancient and have never been updated.

The old files are under "include"  https://github.com/rgleason/weatherfax_pi/tree/master/include
Are these being used?

Here are the new libs
https://github.com/PortAudio/portaudio
https://github.com/PortAudio/portaudio/wiki
https://files.portaudio.com/docs/v19-doxydocs/index.html

filochard, 2024-02-01:
Hi Rick
Building rpms for Linux we need and use the last portaudio sources (our last rpms)
The bundled source in your tree seems to be the headers for Windows (and for Linux when there's no portaudio found on the Linux system) it may be used for flatpak too ...
It won't hurt to update it

filochard, 2024-02-02:
Hi again Rick
the headers you have to update are here :
 https://github.com/PortAudio/portaudio/tree/master/include
You can download them one by one, and then compare them to what you have in you tree 
https://github.com/rgleason/weatherfax_pi/tree/master/include
Then you may do the same thing as you did for libaudiofile (I remember you used winmerge... but I'm not familiar with program since I work only with Linux  for twenty years)

rgleason, 2024-02-02:
Thanks very much filochard, I'll try to do this in the next few days.

rgleason, 2024-12-22:
@filochard 
Hi I've gotten to this task finally.  There seem to be just 7 header files in wxfax and
there are now 11 header files in https://github.com/PortAudio/portaudio/tree/master/include 

Do you think I should just copy them all over and push the change?

rgleason, 2024-12-22:
https://github.com/rgleason/weatherfax_pi/commit/0692830118a54c16d80dbc46cb3618ce50ff63d5

Added all of them.

Closing

closed 2024-12-22 by rgleason

filochard, 2024-12-22:
Hi Rick
As I told you those 4 extra header files are necessary for Linux or Mac  (the 7 already present are only used for Windows) so they are needed if portaudio is not already provided by Linux or MacOS !
(As you may know I continue to use a legacy way to install OpenCPN and its plugins : for weatherfax I never had problem with my rpm package since I use the Linux system portaudio packages)