rgleason/weatherfax_pi #166: Wx XML Update:  Aviation Weather

rgleason, 2024-02-15:
Remove Aviation weather. Website updated and info is no longer useful.
https://github.com/rgleason/weatherfax_pi/issues/157#issuecomment-1941974341

filochard, 2024-02-18:
Hi Rick

To remove Aviationweather
you need to modify a little /src/InternetRetrievalDialog.cpp : 
suppress line 167   _T("Aviation_Weather"),

and when you'll have updated,  with the zip provided by ozolli : 
CoordinateSets-high.xml
CoordinateSets-normal.xml
CoordinateSets.xml

suppress inside them all the lines beginning with : 
<Coordinate Name="Aviation Weather

filochard, 2024-02-19:
Hi again
I forgot to say that you will have also to remove 
WeatherFaxInternetRetrieval_Aviation_Weather.xml 
and 
WeatherFaxInternetRetrieval_Aviation_Weather.xml.bak
from the data folder

rgleason, 2024-12-22:
Thanks for the good checklist. This is done, committed and pushed.
```
To remove Aviationweather
you need to modify a little /src/InternetRetrievalDialog.cpp :
suppress line 167 _T("Aviation_Weather"),

and when you'll have updated, with the zip provided by ozolli :
CoordinateSets-high.xml
CoordinateSets-normal.xml
CoordinateSets.xml

suppress inside them all the lines beginning with :
<Coordinate Name="Aviation Weather
```

Closing

rgleason, 2024-12-22:
I have removed aviationweather because the links did not work.
However the website has changed and I think it would be useful to reinstate it with updated links.

I tested Oliver's revised coordinatesets and aviation weather file and the first urls did not work, but there is good data here.  Look at this chart which shows winds at surface level and then various heights! This is a tremendous resource.

Also the areas of thunderstorms is outlined in brown!  We need this data in weatherfax!
https://aviationweather.gov/gfa/#winds

https://aviationweather.gov/gfa/#winds

filochard, 2024-12-22:
> I have removed aviationweather because the links did not work. However the website has changed and I think it would be useful to reinstate it with updated links.
>
> I tested Oliver's revised [...]

This links doesn't provide downloadable faxes : 
 the link for faxes is always https://aviationweather.gov/fax/

the faxes may be found there but NB the lowest level is always 5 000 feet

rgleason, 2024-12-22:
Yes, I too find upper level faxes.  I wonder where the surface faxes are?
There should be surface faxes because this link shows surface faxes very clearly.
(see the MSL selection on the left side?)
I don't think we are looking in the right place for them.
I guess this is just a work in progress, or it may be shelved!
But I really like the ability to move up through the levels to see how upper level winds affect the surface.  We could learn a lot about surface winds with this tool.

https://aviationweather.gov/gfa/#winds

filochard, 2024-12-23:
> Yes, I too find upper level faxes. I wonder where the surface faxes are? There should be surface faxes because this link shows surface faxes very clearly. (see the MSL selection on the left side?) I [...]

Hi Rick 
this link shows an interactive realtime map for winds surrounding the USA : it is not downloadable
the only way to find faxes is here
[image]

If you click on fax charts you get this : 
https://aviationweather.gov/fax/
There is nothing for sea-level : the lowest level is 5000 feet, and  at this level the wind strength and orientation are not modified by the effect of sea surface (that concerns the sailors : they need to know the wind they will really meet !)

Nevertheless I agree that to get some idea of the forecast these faxes may be someway useful but this need to be explicitly said !
(other faxes sources are good enough and specific for sailors)

In the past this site provided satellites images it doesn't provide them anymore
Other sites provide this kind of images now ...

rgleason, 2024-12-23:
Sadly, Aviation weather does not provide surface level as faxes. However online use of https://aviationweather.gov/gfa/#winds aids in understanding of how upper level winds affect surface winds and locates thunderstorms.

closed 2024-12-23 by rgleason