rgleason/weatherfax_pi #168: Weatherfax plugin will not install or boot since recent updates to Open CPN

Devman25, 2024-09-27:
Recently had it where OpenCPN updated enough that my plugins had disappeared. This was with OpenCPN 5.8.2. Upon reinstalling my plugins - only my WeatherFax plugin had an issue - and said "an error occurred during installation and the plugin will be uninstalled". Every time I open OpenCPN I also get an error saying "The following plugin has encountered an error during startup (WeatherFax) - It will be uninstalled". I've since tried updating OpenCPN further to the latest (OpenCPN 5.10.2) and my operating system (MacOS 15.0) and the problem still exists.

Please let me know of any solutions to this issue - really hoping to be able to use this plugin ASAP.

Thanks!

rgleason, 2024-09-27:
Have updated Logbook's shipdriver template to SD3.3.0 and the metadata and tar are on cloudsmith here  https://cloudsmith.io/~opencpn/repos/logbookkonni-prod/packages/
You can find your OS and download and install, or wait until I push it to the plugins repos.

Devman25, 2024-09-28:
Hi Rick - are you aware of a way to solve the install issue with the Weatherfax plugin?

rgleason, 2024-09-30:
I need to update the frontend first, will do this in several days.

Devman25, 2024-10-01:
Hi Rick,

Thanks for getting back to me and working on it. Hopeful to have it working ASAP as I’m heading offshore shortly and would like to use it for receiving weather forecasts while offshore.

Keep me posted.

Cheers,

-Devin

Devin Jones
***@***.***

rgleason, 2024-10-03:
Have you updated the PLugin Catalog!  
That is a very important first step.

Devman25, 2024-10-03:
Hi Rick,

Thanks for getting back to me.

I have updated the plugin catalog and just updated it again now and tried again. The Weatherfax plugin still will not install properly.

Please let me know what next steps you’d advise.

Thanks,

-Devin

Devin Jones
613-371-6751
***@***.***

rgleason, 2024-10-07:
Weatherfax is working for me on Windows 11.  Do you know how to use it?
Will you please do a search on weatherfax_pi and advise what directories it is found in?

I think your older version may have been installed in a different directory, and you are updating from a very much older version.
You will want to first remove weatherfax plugin from the Options > Plugin page. Close Opencpn
Then do a search on weatherfax  and weatherfax_pi  find the instances of this and remove them.
Then go back to OpenCPN and update the catalog and install the plugin.
Hopefully it will work.

Devman25, 2024-10-21:
Hi Rick,

Thanks for getting back to me.

I’m familiar with how to use the Weatherfax plugin as I used it regularly until it disappeared from my toolbar following a recent update of OpenCPN to 5.10.2. All other plugins have reinstalled and work without issue.

I’m on MacOS Secquoia 15.0.

Tried searching “weatherfax_pi” and my computer didn’t turn up anything. Same goes for “weatherfax”. On the options->plugin page it shows as if it’s unintalled.

Please let me know how to proceed. I’ve attached a screenshot of the error I get each time I open OpenCPN.

Thanks,

-Devin

Devin Jones
613-371-6751
***@***.***

[Screenshot 2024-10-21 at 13.06.48.png]

rgleason, 2024-10-21:
You are using MacOS Secquoia 15.0.  and   OpenCPN to 5.10.2.  
You cannot find Weatherfax 1.9.20.
Have you updated your plugin catalog and installed weatherfax? 
Please refer to the manual for installation of plugins.

Devman25, 2024-10-22:
Hi Rick,

I can find Weatherfax 1.9.20 in the plugin installer - but regardless whether I click the 1.9.111.1 or 1.9.20.0 - neither successfully installs and I get the error that I already sent you whenever I startup OpenCPN. I’ve also tried uninstalling and reinstalling OpenCPN.

Please let me know what to try next.

Thanks,

-Devin

Devin Jones
***@***.***

[Screenshot 2024-10-21 at 17.37.56.png]

rgleason, 2024-10-22:
Thank you for your report. Devman25 for your report and effort. We'll try to get this fixed.

Devman25, 2024-10-22:
Thanks a ton.

Will stay tuned!

-Devin

Devin Jones
***@***.***

Devman25, 2024-11-03:
Hi Rick,

Hope you’ve had a great week.

Was wondering if there has been any progress in fixing the Weatherfax plugin? I’m still having the same issue.

Thanks!

-Devin

Devin Jones
***@***.***

rgleason, 2024-12-20:
@Devman25   We are having some difficulties getting bugs resolved, but I hope this will install on you Mac. Can you please pick the right tarball version and download it and use Import Plugin Tarball to install.  Let me know here please.
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=1.10.3+tarball+mac
Thanks.

rgleason, 2024-12-20:
@Devman25 
Can you also advise regarding these other older MacOS bugs?  Are they fixed or a problem still?
https://github.com/rgleason/weatherfax_pi/issues/112   Can this be closed now?

Devman25, 2024-12-20:
Sorry Rick. Haven’t had time to mess with it as I’m sailing down the coast to Mexico and don’t have starlink or time at the moment. Will reply as soon as I arrive and have a chance. REALLY appreciate your help and efforts!

-Devin

rgleason, 2024-12-20:
Ok thanks.

rgleason, 2024-12-23:
Devman, if you happen to get a chance, theres a new version here 
https://cloudsmith.io/~opencpn/repos/weatherfax-beta/packages/?q=1.10.4+darwin++tarball

Pick the most recent one that matches your MacOS please.

rgleason, 2024-12-23:
Closing, if it does not work. Reopen or post new.

closed 2024-12-23 by rgleason

Devman25, 2025-01-07:
Hi Rick,

Happy new year!

Wanted to report that I’ve finally had a chance to install the new update and it has infact solved the problems I was having.

Thanks a ton for all your help in troubleshooting this and following up!

Cheers,

-Devin