rgleason/weatherfax_pi #173: Problems with ASCAT Indexes being Recognized

rgleason, 2024-12-18:
Recent Windows version almost ready to push to plugins but it does not work  Weatherfax 1.10.3
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=tarball+msvc+wx32+1.10.3

Steps
1. Imported to Opencpn 5.10.2 and enable.
2. Open the menu for weatherfax
3. Select  Retrieve > Update Data Sources
4. Message stating coordinatesets.xml has been saved to coordinatesets.xml.save and replaced by new version.
5. Select  Retrieve > Update Data Sources  again just to be sure.
6. Message stating coordinatesets.xml has been saved to coordinatesets.xml.save and replaced by new version.
7. Select Retrieve > Internet >  Under Servers select ASCAT Indexs  > Under Region select North America or Europe Index (same thing happens)
8. The map appears in the selection window above.
9. Now select the map under "Scheduled" leaving an "X".
10. Next select Scheduled down below, to download the ASCAT Index map.
11. Problem: The map downloads and appears in the Coordinates Wizard rather than overlaying automatically.

I have tried everything I could think of to get this to work properly. I wonder if it is something in the code of the last two entries of the coordinatesets.xml file.  I have run the coordinates wizard and a number of times and 
saved the resulting xml file from  C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml   
 to copy over   C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordinatesets.xml

This file has been pushed up to my repository under the Data directory and is properly being downloaded when the user   Selects  Retrieve > Update Data Sources.

So why doesn't it overlay automatically, requiring the user to operate the CoordinateSets Wizard?

1. Ran the wizard, so the Indexes overlay properly, automatically saving the coordinates to the coordinatesets.xml file in .C:\ProgramData\opencpn\plugins\weatherfax 

[image]
[image]

2. Copied C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml to C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordinatesets.xml

[image]
[image]

Then incremented the weatherfax_pi version number and pushed the plugin code to github, including the data/coordinatesets.xml file

I let Circleci build the various OS and then retrieved the built plugin tarball for Windows from here
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=tarball+msvc+wx32+1.10.3

I downloaded this plugin and imported and enabled it into Opencpn 5.10.2
Then I opened Weatherfax_pi and   Retrieve > Update data sources.  I did this several times to be sure it was loaded.

[image]

Then I it tried the overlay it failed, so I closed weatherfax and then reopened. It still failed to overlay automatically.
Then I tried closing Opencpn and reopening and loading the ASCAT Indexes again and they still failed.

[image]

[image]

Test retrieve ASCAT Indexes
[image]

Coordinates Wizard Opens instead of automatic overlay 
[image]

rgleason, 2024-12-19:
This is just to record an email sent to Dave Burch which has some instructions about using Coodinatesets.xml

**Just download this Windows version   **Weatherfax 1.10.3** (almost ready to push to plugins but it does not work)
https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=tarball+msvc+wx32+1.10.3

and install it using **Options > Plugins > "Import Plugin"** from within OpenCPN.

3. David Burch you don't have to do this below, **just import the plugin** and the new data file coordinatesets.xml should be installed. **Then try the ASCAT indexes overlay**.   --They probably won't work.

4. Then try to use **Retrieve >Update Data Sources**   (twice to be sure)

5. Then **try the ASCAT indexes overlay** again - They probably won't work.

6. Then check the User Data folder **C:\ProgramData\opencpn\plugins\weatherfax\**  it should have two files in it

    **coordinatesets.xml.save** (old version copied here by Update Data Sources)

    **coordinatesets.xml**   (file actually being used by weatherfax_pi and by the coordinates wizard).

    (Note:  Coordinates Wizard records user-made changes to the coordinatesets.xml)

7. In the user data folder ,  the coordinatests.xml should  have an ending showing the ASCAT indexes exactly this way after updating data sources (and also after installing the new version of the plugin) 
   See the image in the comment above for this file.  **(2A)**

I hope this is clear now.   If we need to face time that is possible if the above does not work.

rgleason, 2024-12-19:
OK here is a clue to the problem. When all vestiges of weatherfax_pi are removed and weatherfax_pi  1.10.3 is installed, the ASCAT Index overlays work just fine.

What is happening? Isn't the full and unchanged coordinatesets.xml corrected file being copied from 

- C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordinatesets.xml
- into 
- C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml ?

by the plugin when there are existing files present?  I did check this and it appears that they are being copied but they don't seem to work properly with an automatic overlay.

rgleason, 2024-12-19:
This is a recent email I sent to David who was having difficulties:

David,

This time I decided to start totally clean of weatherfax_pi and did the following:

1. In OpenCPN uninstall the plugin and close Opencpn.
2. Go to  C:\Users\fcgle\AppData\Local\opencpn\plugins  and delete the weatherfax_pi folder
3. Go to C:\ProgramData\opencpn\plugins  and delete the weatherfax folder
4. Go to C:\ProgramData\opencpn\plugins\cache\metadata and delete all the files in that folder.
5. Got to  C:\ProgramData\opencpn\plugins\cache\tarballs  and delete any cached weatherfax  tar.gz files.
6. Start OpenCPN and use Options > Plugins > Import Plugins to import weatherfax 1.10.3

https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/?q=tarball+msvc+wx32+1.10.3

7. Install and enable weatherfax, and open the first menu.
8. Then pick   Servers: ASCAT Index  and  Region: North America Index
9. In the upper window put an X on Scheduled  "ASCAT Indexes North America and hit "Scheduled" below.
10. The index comes up nicely overlaid on Opencpn Charts!!!!
Same for European Index.
Try it. Then we can Zoom if it isnt working.

I think there is something prohibiting the copy of the Data files to the user directory in

    C:\ProgramData\opencpn\plugins\weatherfax  

because it is working now after cleaning up the weatherfax files and starting fresh.

Also all of the ASCAT B files overlay except the Adriatic ones, these two need to be fixed.

I started to check the ASCAT C files but did not get very far.

rgleason, 2024-12-19:
One thing I realize is that when installing the plugin with the corrected xml files, we want to copy the newly installed file

  C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordinatesets.xml

into or over

  C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml

I think David B has been doing the reverse.

rgleason, 2024-12-19:
Response to email from David B.

No they shouldn't have to do anything.

Just install the new plugin with the new data, without having to remove the old plugin and left over parts,  and start using it.  **- This does not seem to be working.**

Maybe they may have to use  "_Retrieve > Update Data Source_" (at the most),   -**This does not seem to be working either**

On 12/19/2024 3:11 PM, David Burch wrote:
> Rick
> That works! 
> Thank you for all the efforts.  
> Now i just stand by to see how you best describe it to typical users to get this going?
> Will they have to do all that i just did?
>
> — david

rgleason, 2024-12-19:
**Observation:** 

When I delete all coordinatesets.xml and saves in

`C:\ProgramData\opencpn\plugins\weatherfax`

and copy and expand this zip file with a version of coordinatesets.xml that works for ASCAT Indexes.  

And then run weatherfax 1.10.3 , Both US and European indexs overlay properly.

rgleason, 2024-12-19:
**Observation:**

When I then run  **Retrieve > Update Data Sources**  and test the ASCAT index overlays, they still overlay properly.

rgleason, 2024-12-19:
Observations:
When I simply delete the coordinatesets.xml file and try the ASCAT Index overlays, they both work, although there is no coordinatesets.xml file in the directory

`C:\ProgramData\opencpn\plugins\weatherfax\`

The coordinatesets.xml file only appears after OpenCPN is shut down.

rgleason, 2024-12-19:
**Observation:**
When I use a text editor on the 

`C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml`

Deleting one of the ASCAT Indexes and saving the file. 
Then when I run "Retrieve > Update Data Sources"  the original file name is changed to  coordinatesets.xml and the new coordinatesets.xml file has both entries for ASCAT indexes down at the bottom.

However when I try an overlay of the ASCAT indexes the one that was not removed works properly, but the other one does not and requires the use of Coordinate Wizard.

rgleason, 2024-12-19:
Observation:
If I then shutdown Opencpn and restart and open weatherfax and try to overlay the two ASCAT indexes, the one that was never deleted overlays properly, the one that was deleted and then restored by using "Retrieve > Update Data Source" fails to overlay properly.

rgleason, 2024-12-20:
Observation: 
If I clear out all files in C:\ProgramData\opencpn\plugins\weatherfax
and  Retrieve > Update Data Files 

The ASCAT index overlays work properly.

===========
If I copy C:\Users\fcgle\AppData\Local\opencpn\plugins\weatherfax_pi\data\coordinatesets.xml

TO  C:\ProgramData\opencpn\plugins\weatherfax

and then remove one of the records for ASCAT Index (say Europe's) using my text editor and save the file.
Then ASCAT Index for Europe does not overlay the chart.  Which is perfectly OK.

Then if I use Retrieve > Update Data Sources  this should fix the problem because the 
C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml should be changed to coordinatesets.xml.save
and both indexes should overlay properly.

Does that happen?

rgleason, 2024-12-20:
Notes

So, that was an example of what the user has to do when an image does not overlay automatically.
They use coordinate wizard and the corrected file is made only when opencpn shuts down.

Would it fix the situation if the wizard wrote the immediately to /user.../loca/...

Yes, it might, but there is something else funny that happens...

I was going to show you that when we use Retrieve > Update Data Source the overlay problem does not get fixed.  When this is done, the data files including coordinatesets.xml from GitHub/rgleason/weatherfax_pi/data should be downloaded to 

%localappdata%\opencpn\plguins\weattherfax_pi\data

and the file in 
C:\ProgramData\opencpn\plugins\weatherfax\coordinatesets.xml ----> renamed to  .save 
This does happen because I find a file like that afterwards,

but I think maybe the  old .xml file still exists ???? maybe?
It would be ok if it just did not exist after a Update Data Sources
because then the coordiantesets.xml in %localappdata%\opencpn\plguins\weattherfax_pi\data
would be used until the USER actually made some change.

OK?
So, if the user does update data sources, all the new info should go into %localappdata%, and any cordinatesets.xml that is in there should first be renamed to .save, also in %localappdata%.  In other words, forget completely about ProgramData ?

Well that is not what I meant really.

Why do we need to use anything from ProgramData?  This location is supposed to be unwriteable by normal use.

Humm. I guess we have it backwards?  Should the program data go to C:\ProgramData\opencpn\plugins\weatherfax?

That is what we had in the "old days".  We changed to localappdata because that is the place where this sort of stuff is supposed to go.  Imagine a multi -user setup.  Each user should have their own data storage.

Yes that is a good point. I guess wxfax needs to get up to that standard.

Yes, we are halfway there.  Needs to be finished.

So, coordinate Wizard needs to write to and from localappdata/opencpn/plugins/weatherfax/data or whatever.

Yes.  And when update data sources is called, first we need to save existing coordinates.xml to ---.save.  This avoids overwriting any local changes made by wizard earlier.

I like it.

bdbcat, 2024-12-21:
Rick...
Sorry, the discussion above is turned around.  My mistake.  Recommend that you delete it to avoid confusion.

bdbcat, 2024-12-21:
We want to store the _dynamic_ data in C:\ProgramData\opencpn\plugins\weatherfax\, which should include all of the data downloaded from the servers.
Everything in localappdata was installed there as part of a plugin installation, and should be considered constant.
I will take a look at the code, and propose corrections.

rgleason, 2024-12-21:
I was thinking about that. Yes I think weatherfax my have its operations on these to folders switched around, and that should be fixed. 

However there is something missing about Update data sources, because after deletion of a record of coordinatesets.xml in (referring to and using the correct user directory) in  %localappdata%/opencpn/plugins/weatherfax_pi

That coord record is missing so the wizard is required.

Then when Update Data Sources is used, the wizard should not be required, because the data sources in c:/programdata/opencpn/plugins/weatherfax/data are updated and the coordinatesets.xml file in %localappdata% is simply renamed to .save. 

Then the coordinatesets.xml in c:/programdata/opencpn... is used until the user creates his own modified version in %localappdata%.

Is that what you are thinking?

rgleason, 2024-12-22:
Dave has solved this problem after an intensive investigation.
He found the merge code for coordinatesets.xml was faulty and fixed it.

closed 2024-12-22 by rgleason