rgleason/weatherfax_pi #183: BUG - MacOS  load new PI,  open OpenCPN, plugin problem and will be uninstalled.  3 different Macs

rgleason, 2024-12-29:
David Burch wrote:
> We just got this comment on the video and i have to agree. The new plugin does not seem to work on Macs. We have tried on 3 different macs.
> After you load the new one, every time you open OpenCPN it  says the plugin has a problem and will be uninstalled.
> Unfortunately, i am tied up for the next week or so and cannot help more with that issue at this time.
> 
[image]

rgleason, 2024-12-31:
@bdbcat @jongough @sebastien-rosset

Also see  BUG: MacOS: segmentation fault while loading the Polar plugin #29 
https://github.com/rgleason/polar_pi/issues/29

Also see VDR  Shipdriver Template  https://github.com/rgleason/vdr_pi/issues/57

rgleason, 2024-12-31:
> I am unaware of wxCurl being used in other plugins. I believe that AISradar is working, OK on MacOS, but I have no confirmation of that yet.   I believe Pavel added wxCurl to weatherfax in order to implement the "Update Data Sources" feature.
> I don't believe other plugins use wxCurl. Is there a way I could search the src folders to check?
> 

> Jon...
> The CI script downloads, builds, and installs openssl to /usr/lib on teh build-bot machine.  At runtime, the plugin assumes that openssl is available on the target machine.
> I am not completely sure how we ensure that this is true.
> The openssl dependency come from wxCurl, which depends on libcurl.  I wonder what other plugins use wxCurl, and how this is handled there.
> Dave

> Hi Dave, Rick,
>    I have just installed OCPN 5.10.2 on macOS after giving permission to the do the install. It seems to work OK with weatherfax. So it is just the missing openssl that is the problem.
> Regards    Jon
> 

> Hi Dave, Rick,
>    On my macOS ventura there is a '/usr/lib' but no '/usr/local/lib' so that wouldn't help. I have looked in the lib file and there is no libssl.3.dylib file. However, if I install brew and the use that to install openssl the missing file is found in '/usr/local/lib' .
> 
>    To install brew issue this command from a terminal:
> 
>     bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"
> 
> as a standard user and enter your password when requested. Then you can just use 'brew install openssl" as a user, again you may be asked for the password. Then weatherfax will install and work, I think, within OCPN. I am not sure how you can automate that in plugin manager, but....
> 
> 
> Regards
>    Jon

On 30/12/24 14:39, Dave wrote:

> All...
> And there you have it:
> Library not loaded: /usr/local/lib/libssl.3.dylib
> 
> This looks like a linux file location specifier.  Probably not right for MacOS.  CMakeLists need a tweak for this platform, I guess.  Will need to compare with other plugins that do network comms, and are also built using this template.
> 
> Dave
> 

On 12/29/24 22:27, Jon Gough wrote:
> Hi Rick,
>    I have now managed to get macOS ventura-13 running and it has OCPN 5.8.4-0 installed and it updated OD to 1.8.58.0 successfully. I got AISradar_pi 1.4.18 installed but weatherfax_pi 1.10.6.0 nor 1.9.111.1 will run. Looking in the log I see this:
>
>     14:10:48.420   ERROR dlunix.cpp:120 Failed to load shared library '/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib': dlopen(/Users/jon/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib, 0x0002): Library not loaded: /usr/local/lib/libssl.3.dylib
>
> Hope that helps.
> Regards
>    Jon

bdbcat, 2024-12-31:
Rick...
Now involving Alec, who wrote the libcurl CMake scripts.
We are beginning to understand the issue, now looking for proper solution. We shall see.  I expect the solution will appear as a change to the opecpn-libs submodule, and not to WFAX itself.

It appears that we never had a version of WFAX for MacOS that used curl to update the data sources.  Does that sound right to you?

rgleason, 2024-12-31:
Thanks.
I am very hazy on old MacOS issues.  I don't believe we got many reports about wxfax from Mac users about this topic, so it was missed. Therefore probably not many/any  MacOS users of wxfax?   They just avoided it? Short- I don't know.

Is this PR still current?
https://github.com/rgleason/weatherfax_pi/pull/184

bdbcat, 2024-12-31:
#184 is ready.

bdbcat, 2024-12-31:
Australian BOM tested OK on Windows.

rgleason, 2025-01-01:
Dave

I will push but it is missing 2 files. Then will build again with new version number.
We had about 3 x2 file synch failures on Cloudsmith. 
The message said there was a duplicate file
and to delete. I did but there there wasn't a dup file.  
See https://cloudsmith.io/~opencpn/repos/weatherfax-prod/packages/
It is the files with the little red dot.

Only 15 files. Missing 2
        metadata/weatherfax_pi-1.10.7.0-android-arm64-16-android-arm64.xml
        metadata/weatherfax_pi-1.10.7.0-android-armhf-16-android-armhf.xml
        metadata/weatherfax_pi-1.10.7.0-darwin-wx315-x86_64-12.6.3-macos.xml
        metadata/weatherfax_pi-1.10.7.0-darwin-wx32-arm64-x86_64-12.6.3-macos.xml
        metadata/weatherfax_pi-1.10.7.0-debian-arm64-11-bullseye-arm64.xml

        metadata/weatherfax_pi-1.10.7.0-debian-arm64-12-bookworm-arm64.xml
        metadata/weatherfax_pi-1.10.7.0-debian-armhf-11-bullseye-armhf.xml
        metadata/weatherfax_pi-1.10.7.0-debian-armhf-12-bookworm-armhf.xml
        metadata/weatherfax_pi-1.10.7.0-debian-x86_64-11-bullseye.xml
        metadata/weatherfax_pi-1.10.7.0-debian-x86_64-12-bookworm.xml

        metadata/weatherfax_pi-1.10.7.0-flatpak-aarch64-22.08-flatpak.xml
        metadata/weatherfax_pi-1.10.7.0-flatpak-aarch64-24.08-flatpak.xml
        metadata/weatherfax_pi-1.10.7.0-flatpak-x86_64-24.08-flatpak.xml
        metadata/weatherfax_pi-1.10.7.0-msvc-x86-wx32-10.0.20348-MSVC.xml
        metadata/weatherfax_pi-1.10.7.0-ubuntu-x86_64-22.04-jammy.xml

Missing
  metadata/weatherfax_pi-1.10.7.0-msvc-x86-10.0.20348-MSVC.xml
metadata/weatherfax_pi-1.10.7.0-flatpak-x86_64-22.08-flatpak.xml

bdbcat, 2025-01-01:
"Missing 2" push PR failed, not merged.

rgleason, 2025-01-01:
Yes, sadly. Will increment and publish tomorrow morning.  Happy New Year!  Headed to bed.

rgleason, 2025-01-12:
Resolved. Closing.  Now there is a problem with the size of the second menu which opens closed and users don't understand that it needs to be stretched wider and higher. They think the PI is not working. Will add an issue.

closed 2025-01-12 by rgleason